motioneye-project/motioneyeos #2294: Not recording when not connected to network

iliesaya, 2020-02-17:
Motion detection is working fine, but only when raspberry pi is connected to the network. 
wifi not configured. 
When RJ45 cable is connected, I can access to motionEye and camera record motion, but when I unplug the cable it's not recording anything. 
Did I miss something ?

[boot.log](https://github.com/ccrisan/motioneyeos/files/4212797/boot.log)
[dmesg (1).log](https://github.com/ccrisan/motioneyeos/files/4212798/dmesg.1.log)
[motioneye-config.tar.gz](https://github.com/ccrisan/motioneyeos/files/4212792/motioneye-config.tar.gz)
[motion (1).log](https://github.com/ccrisan/motioneyeos/files/4212794/motion.1.log)
[motioneye (1).log](https://github.com/ccrisan/motioneyeos/files/4212795/motioneye.1.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/4212796/messages.log)

starbasessd, 2020-02-17:
Which PI?
Have you tried upgrading to 20200203?
If you have a monitor connected, and you unplug the Ethernet, does it
immediately start rebooting in a loop?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-02-17:
I am unable to find dev20200127 anywhere. I would suggest upgrading to the
latest stable version:20200203 or downgrading to 20190911
In the mean time, I will test on RPi3B+, but would like to know which PI
you are using.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

iliesaya, 2020-02-17:
It's a raspberry pi 3B+ , I plugged a screen and yes it is in fact rebooting every 1 min.
is there already a fix or should I upgrade ?

starbasessd, 2020-02-17:
1) You should upgrade, regardless.
2) Let me check a few things before you doing so.
There is an issue with networking, that if you don't change a configuration
file, when the Pi doesn't see a network, ethernet or wifi, it goes into the
boot loop.
Q: Would you like to be able to access the PI when not connected to
ethernet? It doesn't matter if you connect to a home wifi, it can be it's
own access point. You could access via a phone or tablet, and view or
download recordings, and view the live camera.
This is a fully supported feature, not a 'hack'.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

iliesaya, 2020-02-17:
I didn't know such a feature existed, yes, that would be useful :)

starbasessd, 2020-02-17:
MotionEyeOS Stand-alone Setup
MotionEye has the ability to act as a stand-alone WiFi IP camera, even in a non-networked environment. 
To do so, you need a Raspberry Pi, and MicroSD card, along with a PC to image the SD Card. If you do not have a Pi 3, 3B+, 4B,
or Zero W or Zero WH, you will also need a supported USB WiFi adapter.
Download the latestest stable image of motionEyeOS for the Raspberry Pi you will be using. (as of 17 Feb 2020 it is
20200203)  https://github.com/ccrisan/motioneyeos/releases
Image the SD Card using the tool of your choice. I recommend Win32DiskImager, but Etcher also works in Windows.
Do not remove the SD Card yet. You need to create or copy the following files to the /boot drive:
hostapd.conf , dnsmasq.conf , os.conf , and static_ip.conf . I will include the contents at the end of this tutorial.
If you are using Windows, do not create/edit these files with notepad or wordpad. They create files that aren't quite correct for Linux. I recommend Notepad++, but you can use anything that supports Linux files.
Eject the SD Card, and place it in your Pi. If you are going to use a USB or CSI Camera, attach it. 
Power on the Pi. 
Wait about 2-3 minutes for the first boot and setup.
On your Android Phone, turn on WiFi and connect to 'picam' (if using the settings below in hostapd.conf)
Open the web browser on the phone and go to http://192.168.42.1 (again, if using the settings below in static_ip.conf)
You should get a login. Username is 'admin', password is blank.
You should now be at motionEye screen, with a link to add a camera.
All the normal setup configurations will now work as in the instructions at: https://github.com/ccrisan/motioneyeos/wiki/Configuration
Congratulations!
If you would like more assistance, please put in a ticket at:
https://kevinshumaker.wixsite.com/thethirdlevel

Files needed in /boot before first startup:
Filename:	os.conf
========
OS_NETWORKLESS="true"
========
--------
Filename:	static_ip.conf
========
STATIC_IP="192.168.42.1/24"
STATIC_GW="192.168.42.1"
STATIC_DNS="192.168.42.1"
========
--------
Filename:	hostapd.conf
========
ssid=picam
wpa_passphrase=12345678
wpa=2
wpa_key_mgmt=WPA-PSK
interface=wlan0
channel=1
driver=nl80211
hw_mode=g
ieee80211n=1
ieee80211ac=1
========
--------
Filename:	dnsmasq.conf
========
interface=wlan0
dhcp-range=192.168.42.50,192.168.42.150,24h
dhcp-leasefile=/data/dnsmasq.leases
========

[ConfigFiles.zip](https://github.com/ccrisan/motioneyeos/files/4213645/ConfigFiles.zip)

starbasessd, 2020-02-17:
BTW, this effectively disables the ethernet connection, so you won't be able to view or set up from a PC or laptop, unless it has wifi and can connect to the picam wifi.
All other configuration works as before.

iliesaya, 2020-02-17:
> BTW, this effectively disables the ethernet connection, so you won't be able to view or set up from a PC or laptop, unless it has wifi and can connect to the picam wifi.
> All other configuration works as before.

oh, I understand, that's a problem because my laptop is connecting with wifi so I can't be connected to both my metwork and the rpi. 

I updated to the latest 20200203 , does it still have the rebooting issue when not connected ?

starbasessd, 2020-02-17:
Yes.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-02-17:
Do you have a cell phone or tablet? You can use them to connect to the Pi.
Or you can use the laptop as an intermediary.
Why don't you want the motionEye connecting to your WiFi, if I may ask?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

iliesaya, 2020-02-17:
I am using it in my campus university , which is using a wifi feature called rogue containment ( https://www.cisco.com/c/en/us/support/docs/wireless/4400-series-wireless-lan-controllers/112045-handling-rogue-cuwn-00.html ) which is killing all other wifi except the "official" campus wifi

starbasessd, 2020-02-17:
Interesting.
Where / when / how would you be viewing the recordings?
What is the camera protecting?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

iliesaya, 2020-02-17:
just my desk in an open office, where other students like to "borrow" my stuff ...

starbasessd, 2020-02-17:
Where / when / how would you be viewing the recordings?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

iliesaya, 2020-02-17:
where : in my same office, I can plug it to the ethernet network of campus
when : maybe every 3 days
how : on my laptop which is connected to the same campus network but with wifi

starbasessd, 2020-02-17:
I'll look and see. Interesting issue.
May take me a bit to figure out. First thought is to turn the
OS_NETWORKLESS="true" to "false" whenever you wanted to use ethernet.
Problematic, though. How to access the os.conf file when network operations
are wanted.

Q: Can you use MiFi devices on campus? (WiFi to Cell networks? Or can you
use a cell phone as a HotSpot for internet access?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

iliesaya, 2020-02-17:
Thx for your help and prompt response, appreciate it :)
about the mifi, it's more or less working but not stable at all, as soon as campus wifi router detect a new SSID they start bombarding it and that make connection drop after a short period.

starbasessd, 2020-02-17:
OK. remind me to never go to your campus.
I've worked in hospital and banking environments, even been on a few
government installations, and no-one goes to that level of interference.
Wow.

Will get back to you as soon as I can.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Luke752, 2020-02-18:
Same problem with PI3b (I haven't now the Meyeos version, a 2019 one) with picam csi. first configuration was done via ethernet then I configured wifi to connect my smartphone as hotspot. after several try I guess Meye don't record if isn't connected with network, I plugged a monitor so I can see continuous rebooting with no network detected. so, if I understand correctly, I must upgrade with last version but isn't enough so I should copy the files in /boot and do the first boot just only with wifi/smartphone without ethernet ? Thank you so much, and sorry I'm a novice

starbasessd, 2020-02-18:
That is correct.
Image the SDCard (preferably with latest 20200203)
Add the 4 files to /boot
Insert in the Pi
Boot it up. If you are watching it on a monitor, wait for the
meye<whatever> login prompt
Use your phone to connect to the pi-cam wifi with password 12345678 if you
didn't change the files.
Use the browser in the phone to go to http://192.168.42.1 if you didn't
change anything in the files.
Log into the Web GUI with admin and no password
DO NOT TOUCH ANYTHING in Networking.
Ethernet is disabled when set as a WiFi Access Point.
You cannot set up the WiFi access point from within the menus.
It will record in the default File Storage location.
From the WebGUI, you can click on the video feed, and some icons will
appear above and below the video.
You can use the
[image: image.png]
2nd from the left to see pics recorded (looks like a picture), or 2nd from
right (Looks like the >) st the top.
You can also download them to the phone, or to a PC connected to the pi-cam
(same as phone)

Let me know if you need help, or get it working.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Luke752, 2020-02-18:
Tomorrow I'll follow everything step by step and I'll report result. Thank you

Luke752, 2020-02-19:
I try to copy 4 files in /boot folder but the error is "permission denied". I'm using raspbian to copy the files, what I'm doing wrong ? Thank you

Luke752, 2020-02-19:
everything works ! Thank you !

starbasessd, 2020-02-19:
You are welcome. Enjoy!

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

2hot2touch, 2021-01-06:
Hello! 
I've just read all Your topics above, but I'm still having similar issue. 
Maybe the presented walkaround may work, like in case with adding 4 files etc., as starbasessd suggested... but what if I don't want to use it in any way with wifi/ethernet connection (beside creating a set up at home)? Setting it up as access point will drain my batteries more than I want, so that's my I'm writing this topic here and looking for help.

The system I've set up is with using RP Zero (no wifi version). I've bought this one, becasue I don't wat to use that feature. I'd like to have it as a wildlife cam. As I've tested at home, I can make it work with saving all the stuff on a flash drive (usb), but when it's not connected to the network it stopped saving anything on the usb flash drive. The step with starting it up without network connection is already done - works ok. 

Working with RPi was planned as visiting the working spot once per week, to change batteries and flash drive, then leave it for a next week. So no connection is required.

When preparing it I use MicroUSB to female USB cable, then usb to RJ45 adapter - works like a charm.

I'd be more than happy if anyone can help me with this one :) Thanks in advance!

starbasessd, 2021-01-06:
@2hot2touch
What is YOUR setup?Which Pi, which version of motionEyeOS?
If your version of motionEye)S is 20200606, please switch to 20190911 or
dev20201026 and let me know if you're still having the issue.
If you don't want any networking, the only file you need to add to the
/boot partition prior to first boot is os.conf with the single line
OS_NETWORKLESS="true"

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

2hot2touch, 2021-01-06:
Thanks for such a rapid answear. 
I use:
-Rasp Pi Zero v 1.3 (no Wifi)
-MotionEyeOS: 20200606 (I will try the one as You've suggested)

As said before: "The step with starting it up without network connection is already done - works ok", so it starts as it should - the change was done and it starts well.

Right now I'm out of reach this RP to try with other version, as You have said - will do ti tomorrow and I'll let You know.

Thank You! :)

starbasessd, 2021-01-06:
20200606 has a bunch of weird quirks, which is why I automatically
recommend 20190911 or dev20201026 out of hand.
I do have a couple of PiZeros (no W or WH) to try if yours fails.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

2hot2touch, 2021-01-07:
Hi!
I've tried it in three different ways - without success...
After flashing a card - it works well with ethernet cable plugged in. Additional file was created as said also.
Without established connection - it didn't save any footage.
1. I've tried to disconnect ethernet cable while all systems were running - no success, no footage taken after disconnecting.
2. I've tried to start it without cable plugged in - no success, no footage taken from the start.
3. I've tried to change a destination directory to exclude a problem with saving footage on USB drive. I've switched it to internal memory - no success, no footage taken while tested in the same way as in listed 1 and 2.

If You would like to reach any particular data to help me with it - just let me know. I'd be more than glad to help You with solving it. Thanks in advance! Best Regards, Mark

starbasessd, 2021-01-07:
Have you tried the dev20201026?
How are you connecting ethernet to a PiZero?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-01-07:
BTW, I typo'd the os.conf line, it should be
OS_NETWORKLESS="true"
I have it up, but now digging through my stuff to find a USB<>Ethernet
adapter, otherwise there is no way to set it up...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-01-07:
Another issue: once os.conf with no networking is installed, guess what? No network, no ethernet, nada. The only way to get into it again without flashing is to use a console cable.

2hot2touch, 2021-01-08:
Hi!
I've tried it only with 20190911 so far.
The problem with boot occures on my RP Zero too. As I've previously tried - it may be passed by editing like that:
"For anyone with same issue create 'watch.conf' file in the root directory of the sd card and enter 'WATCH_LINK="false"'".
That worked for the latest version.

(source: [https://www.gitmemory.com/issue/ccrisan/motioneyeos/2178/548449563](url))

But when edited like that, on latest version, it doesn't save the image. 
I'll check it on 20190911 now, with saving on MicroSD card.
If failed, I'l check with dev20201026.

I'm connecting to Pi via OTG USB<-->MicroUSB cable, then using USB<-->Ethernet network usb adapter (from Digitus company)
[https://allegro.pl/oferta/lan-usb2-0-karta-sieciowa-10-100mb-s-digitus-8521312612](url)

To be honest - I do not know what setting WATCH_LINK to false does, but it allowes to move forward during booting.

starbasessd, 2021-01-08:
Do you mean the watch.conf disappears? It should. The system will read it,
and move the info to the correct location, then delete it out of /boot
Setting to false tells the system to not be continuously checking to see if
you have a network connection.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

2hot2touch, 2021-01-08:
Okay, so it works now! 
It's set up on 20190911, data saves on MicroSD card.
So only changes, from the start, are:
-switching version to 20190911
-changing watch_link to false (as said above)

Now, I need a RTC, because all stuff was named weith wrong timestamp (1970 etc.). ;) 

For now I'm not trying case scenario with the flash drive, because saving on a MicroSD dropped a required Amps by a little, so it can last longer on batteries.

Thanks a lot for Your help! :)

closed 2022-01-19 by starbasessd