motioneye-project/motioneyeos #2295: Can't upload to Google Photo

andreluisos, 2020-02-17:
Accessing the upload service failed: Quota exceed for quota metric 'all requests' and limit 'all requests per day' of services 'photoslibrary.googleapis.com' for consumer 'project_number:..."

That's the error I get when I try to test the key. Tried 3 different accounts, one of them unlimited.

starbasessd, 2020-02-18:
Which version of motionEyeOS?
Which version of Pi?

andreluisos, 2020-02-18:
Sorry for the missing information. I'm using the pre-release version 20200203 on a Raspberry Pi B+.

starbasessd, 2020-02-18:
Re-created with:
Pi3B, 8GB SDCard, Image 20200203
Installed wpa_supplicant.conf
Installed SDCard to Pi3B
Installed ethernet connection (belt & suspenders)
Booted
Accessed Web GUI
Connected to simple mjpeg camera
Logged in as admin
Went to File Storage
Enabled Upload Media files
Selected Google Drive for Upload Service
set location to pi-cam
turned on Include Subfolders
Turned on Clean Cloud
Selected Obtain Key
went through Google steps to create key, copied
Pasted Authorization Key
clicked Apply
Then tested
Success
Enabled movies and pictures
went to camera screen
clicked several times on blue camera icon in lower right
Went to GDrive, folder pi-cam was there, opened, and sub folder today's
date was there, pictures in that folder as expected.

Keys are One Time Use. If anything fails, you must get a new one. Always
"Apply" before testing.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

andreluisos, 2020-02-19:
I've done as you told me to: got a new key, applied the settings and then I tested it, but it failed, the same way as before.

starbasessd, 2020-02-19:
Did you start over, with the clean image and instructions? Did you grab
20200203?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-02-19:
Be sure to grab the latest version, not the pre-release version.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

neildotwilliams, 2020-06-25:
I think as having issues with this. It doesn’t like spaces or underscore in the path for me. I now use /houserear and it works great ;-)