motioneye-project/motioneyeos #2296: pi3 WiFi stops working after 1 month DHCP?

cpc005, 2020-02-17:
Pi3 with motion eye is connected to router via WiFi. The router is provided by TalkTalk UK broadband supplier.
When I initially setup then I log into the router to see what IP has been allocated. In this case it was 192.168.1.12 I can access motioneye interface using above address and it seems OK for a month.

After approx 1 month I find I cannot access motioneye from 192.168.1.12 so perhaps router has allocated it a new address. I log into router and find that pi 3/meye is only listed by its mac address and no IP address is given. If I power cycle the pi3 then I can see from router log that it is seeing the pi3 somehow presumably because of mac broadcast.

I scanned the network for all ip addresses using the 'Fink' scanner - no pi3 has an ip address.

Looking through the attached messages log we can see that  meye thinks  it is getting new IP assigned at this point. Not sure why IP changes at this point 
Feb 11 02:19:22 meye-d052c190 daemon.info dhclient: bound to 192.168.1.27 -- renewal in 42887 seconds.

Perhaps something is going wrong with this process so the IP is not correctly assigned??

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.
yes
I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.
yes
### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

20190911

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).
PI 3B+
### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

MMAL ??

My camera model is: (insert your camera model, e.g. Logitech C290).

PI camera
### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

WiFi
### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

[messages.txt](https://github.com/ccrisan/motioneyeos/files/4215687/messages.txt)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/4215697/dmesg.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/4215698/boot.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/4215699/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/4215700/motioneye.log)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

starbasessd, 2020-02-17:
Under Expert Settings, can you turn on:
Network Link Watch (I set to 1 minute for my Pis)
Connectivity Watch (I set mine to 1 minute)
I also do not use Fast Network Camera.
Do you have it connect to a hub running motionEye or motionEyeOS or is it
stand alone?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

cpc005, 2020-02-18:
It is a stand alone pi with pi camera.

starbasessd, 2020-02-18:
Please try the previous settings. I realize it may take another month, but
I'm willing to wait...
Does any other device exhibit these symptoms? If you have a desktop or
laptop, can you start it at the same time as the PI and see?
Just trying to determine if it is the TalkTalk router or the Pi...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?