motioneye-project/motioneyeos #2298: Motioneyeos - Openmediavault

bobster316, 2020-02-18:
I have two raspberry pi 4's, Desktop PC, all connected on the same lan.

1. MotioneyeOS - RPi4 (CCTV)
2. Openmediavault - RPi4 (NAS - External Hard Disk)

Each on their own are working okay, I am having problems setting up
network shares/folders for the cctv videos to be saved  onto the NAS
external hard drive,.

How do i configure the storage device/network share so i can save videos from motioneyeos to the external hard drive,

starbasessd, 2020-02-18:
Which version of OMV and which version of MotionEyeOS?
I am going to try to reproduce both ends, but I only have 1 RPI4 (4GB).
1st attempt will be OMV5.0.5 on a VirtualBox machine and motionEyeOS
20200203 on the Pi4. If I get that working
I'll reverse it to a P3B+ for motionEyeOS (same version, and the 'beta' 5
version of OMV on the Pi4.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-02-18:
What services do you have set up in OMV?
Have you asked for assistance in OMV forums for the service you want to set
up?
Are you able to save / retrieve files using Windows to the OMV NAS?
OMV is much more involved than anything I've done before just to do NAS. I
use an Ubuntu 18.04 with SMB Services, SFTP services, Serviio, miniDLNA,
and VirtualBox, and WebMIN for administration. I also have a Buffalo
LV-W927 NAS running Apple, FTP, SMB, and DLNA. I do RPiZero Wireless as
mobile media servers for my grandkids. I'm sitting with 4 RPis netbooting
off of an RPi2 w/1TB drive for a school teacher friend of mine around me.
None are as difficult to set up as this is.
Please answer the questions from both replies, if you would.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-02-19:
OK, I've got good news, and I've got bad news...
I can successfully share a drive in OMV to FTP, and access it from multiple
machines including 2 RPis running Raspbian 20200205, an Ubuntu 18.04 and a
Windows 10 Pro. It's not the OMV, I don't think, although the setup is
crazy. More later on that.

I am unable to get motionEyeOS to save to ANY ftp server my FTP server on
my Buffalo NAS, my FTP server in my VirtualBox, or my FTP server on my
Windows box. Sometimes I can run test, and it says 'successful', but never
really works. I had less issues connecting to my GDrive earlier today.

I am at a loss for you, other than put your log files in the ticket.

I'm going to try to figure out the FTP issue, but I don't think it's us.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

bobster316, 2020-02-19:
Thanks for your help, just some additional information regarding the setup

[Raspberry Pi 4 Setup.docx](https://github.com/ccrisan/motioneyeos/files/4224478/Raspberry.Pi.4.Setup.docx)

starbasessd, 2020-02-19:
Nice cpu coolers. Linus of LTT would be proud of all the RGB!

If I am seeing this correctly, the settings for File Storage are incorrect.
Also the locally stored video seems to have filled the local drive.

My process for testing:
Create a new imaged SDCard (8GB, Class 10)
Using image 20200203
copy wpa_supplicant.conf to /boot
install in PI
Boot normally Ethernet and WiFi connected
Connect Network MJPEG camera via Web GUI
On FTP NAS Server (Buffalo LS-WVL972
Create new user motion & password, give full r/w access
On motionEye PI go to Settings, File Storage
Leave Storage Device at defaults
Enable Upload Media Files
Upload Service set to FTP Server
Server Address set to 192.168.2.252
Server Port set to 21
Location set to /disk1/Share_Drive1/motion/
Enable Include Subfolders
enter username motion & password
Click Apply at the top before testing
Click Test Service
Go down to Still Images, enable
Enable Manual Snapshots
Apply
Switch to video feed
click on video feed.
Click blue camera in lower right, several times.
Check via your favorite FTP Program (I use Firefox to
ftp://192.168.2.252/disk1/Share_Drivew1/motion, and username motion and
password)
<D@MN> it worked.
=====
Will be back
Going back to the OMV to re-set it up.
Watch this space.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-02-19:
Here's a screenshot of the working setup to my Buffalo NAS FTP:
[image: image.png]

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?