motioneye-project/motioneyeos #2301: low fps everything but .avi

zubzero689, 2020-02-22:
[boot.log](https://github.com/ccrisan/motioneyeos/files/4240056/boot.log)
[motioneye.log.txt](https://github.com/ccrisan/motioneyeos/files/4240060/motioneye.log.txt)

I am running motionEyeOS version: 20190911

### Board Model

I am using the following board/model: xu4

### Camera

I am using the following type of camera:  Fast Network Camera and Simple MJPEG Camera x2
both are pi zero w

My camera model is: pi camera 

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

  External Hard Disk ssd formated to ntfs

### Log Files

I consider the following log files relevant to this issue:
 
and yea i have tried everything i can think of and i even tried an external ssd so it could write more data faster then a sd card.  i havent tried the nightly build with the ssd yet i didnt like the nightly last time on the xu4 for me i had to unplug the power after every change that gets bad quick wonder if anyone can help my goal is 720p at 10 fps on .h264 i hardly get 5fps mostly 4.6fps with a high kbps 
or if i could get .avi to record at aleast 1mbps instead of 400kbps bit rate would be nice

jasaw, 2020-02-23:
The latest prerelease has hardware accelerated video encoding support, v4l2m2m, but you are right that it is not able to reboot for some reason. I don't own an XU4, so I can't debug the problem. @ccrisan did you look into this problem?

zubzero689, 2020-02-24:
i will give it another try

zubzero689, 2020-02-24:
i grabbed the dev20200223 release and tied the hardware .mp4 version starts out looking good the just pixals after 2 sec not worth it and i still have to unplug it every time you need to reboot. all the setting saved so thats good it like it tries to boot and stops the alive light stops then thats when i unplug this is also an issue with my pi3b+ i just had a xu4 and know thats better thats why im using it instead of pi3 
more details:  camera are running at 1024x768 10fps acording the software and the records are at 1024x768 4.67fps bitrate 6066kbps or less

jasaw, 2020-02-24:
@zubzero689 Which video encoder have you selected? Did you choose "v4l2m2m_h264" variant?

zubzero689, 2020-02-24:
i did

ccrisan, 2020-02-24:
> ...but you are right that it is not able to reboot for some reason.

It's a known issue and I can't for the life of me understand what's going on. We're using their default kernel source tree and configuration and there's nothing special about the XU4 variant that could cause this, as far as I know.

Also I don't understand why the watchdog doesn't do its job.

zubzero689, 2020-02-24:
i was also wondering is there a way to increase or decrease the bitrate or set it that might help.  one more thing is i dont know what the bottleneck is the the cpu encoding are the high writes the files need to do