motioneye-project/motioneyeos #2304: Pi 3B+ over a 3B?

jerryfudd, 2020-02-25:
Hi,

Not an issue, just a question - I'm running in fast cam mode using a Pi camera on a Pi 3B over cabled ethernet at 1920x1080 - is there any advantage at all in switching to a 3B+?

I have both spare so no financial implication but I'd also rather not use the 3B+ if it will make no difference.

thanks,

Dan

starbasessd, 2020-02-25:
Maybe. Ethernet might be faster if you have Gigabit switches and your other
hardware can use it.
CPU is only marginally faster.
What are you using as your Hub/viewer?

*Spec* *Raspberry Pi 3 B* *Raspberry Pi 3 B+*
*CPU type/speed* ARM Cortex-A53 1.2GHz ARM Cortex-A53 1.4GHz
*RAM size* 1GB SRAM 1GB SRAM
*Integrated Wi-Fi* 2.4GHz 2.4GHz and 5GHz
*Ethernet speed* 10/100 Mbps 300Mbps
*PoE* No Yes
*Bluetooth* 4.1 4.2

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jerryfudd, 2020-02-25:
I'm just viewing over the admin page as I'm the only one who uses it. There is no records, just a live stream.

The ethernet port made me curious but wasn't sure if there was a chance it was saturating the the 3B and the 3B+ is only meant to be marginally faster as its a gigabit port but using the same bus so not gigabit speeds.

Dan

starbasessd, 2020-02-26:
There is a significant difference in speed on ethernet. Theoretical is 3x+,
but like this testing showed, I get 1.5 to 2.5 times speed increase.

https://core-electronics.com.au/tutorials/raspberry-pi-3-model-b-plus-performance-vs-3-model-b.html

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-02-26:
This is depending on your switch, though. If your network switch only supports 10/100, you won't see any improvement. I use all Gigabit, with a fiber backbone in my home network.

jerryfudd, 2020-02-26:
just cat6 here but they are dumb gigabit switches direct back to the main router which is also gigabit.

Dan

starbasessd, 2020-02-26:
I'd just go ahead and test. See if it makes any difference in your
situation, or if the difference is worth it to you.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?