motioneye-project/motioneyeos #2306: Not connecting to wifi

TR-SLimey, 2020-02-26:
### motionEyeOS Version

I am running motionEyeOS version: 20190911.

### Board Model

I am using the following board/model: Raspberry Pi 1 B.

### Camera

I am using the following type of camera: V4L2 (PS3 Camera).

### Network Connection

My motionEyeOS unit is connected to the network via: Nothing. That's why I'm here.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 *  WiFi Dongle (Works with raspbian etc.)

### Log Files

I consider the following log files relevant to this issue:
[boot.log](https://github.com/ccrisan/motioneyeos/files/4258347/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/4258348/dmesg.log)
[messages](https://github.com/ccrisan/motioneyeos/files/4258349/messages.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/4258353/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/4258354/motioneye.log)
[wpa_supplicant.log](https://github.com/ccrisan/motioneyeos/files/4258355/wpa_supplicant.log)

### Issue

MotionEyeOS isn't connecting to my WiFi network though the ethernet connection works and the WiFi credentials are correct. I attach all of the logs.

starbasessd, 2020-02-26:
Using the following wpa_supplicant.conf, copied to /boot before first boot,
connects to my wifi using an EDIMax (8192cu) and Xerox.

-----
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
    ssid="SSID"
    psk="password"
key_mgmt=WPA-PSK
}
-----

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-02-26:
boot log shows:
* Starting wpa_supplicant: no device
 * Configuring wireless network: no device
 * Configuring wired network: dhcp
 * Setting current date using http: Tue Feb 25 20:08:44 UTC 2020
doesn't see wlan0
Sees and connects to ethernet eth0
next to last boot log entries do show:it finally set up wifi: rt2800usb
 * Starting wpa_supplicant: done (rt2800usb)
 * Configuring wireless network: dhcp
but doesn't show an ip address just the gateway.
Some problems showing for various versions of rt2800usb here:
https://elinux.org/RPi_USB_Wi-Fi_Adapters
(I did a find for the name rt2800usb. Some require higher power power
supplies or powered hubs, others need modifications.
I do have a Tenda that reports as an rt2800usb, and is working fine using
my wpa_supplicant.conf file posted previously.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

TR-SLimey, 2020-02-27:
So I've tried 2 different WiFi dongles (both of which work with Linux), tried configuring and not configuring a static IP, tried @starbasessd 's WiFi configuration and nothing so far.

starbasessd, 2020-02-27:
If you want a static IP that is a different config file.
You changed the SSID and Password in the setup I gave you to your SSID and
password?
May I ask your SSID? If it has non-alphanumeric characters (other than the
'-') it may not work correctly.
Are you using NotePad or NotePad++ to edit the files before copying them to
/boot?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

TR-SLimey, 2020-02-27:
I don't really want a static IP, I just set one in case the problem was anything to do with DHCP.

And yes, I did change the SSID and password in the file.

My SSID has an underscore (`_`). Does that make much of a difference? It seems to work everywhere else.

And I'm using GNU/Nano and Gedit to edit the file.

starbasessd, 2020-02-27:
Have you tried a different power supply?
Or plug it into a USB port on a PC or laptop for power.
Also, please try the 20200203 version
https://github.com/ccrisan/motioneyeos/releases/download/20200203/motioneyeos-raspberrypi-20200203.img.xz
I've tested the Tenda rt2800usb on all levels of PI I have, and it works
with all them. PiZero & PiZeroW, Pi1, Pi2, Pi3B and 3B+, and Pi4 using the
correct version of motionEyeOS for each.
As another test, install Raspbian 20200213, and follow the instructions VV
here VV it install motionEye, and see if that works.
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
This will confirm if it is an issue with the version of the OS or
motionEyeOS not working with it. I was able to get a PI1 &3B and the Tenda
works there, too.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

TR-SLimey, 2020-02-29:
Excuse the delay. Was busy. I will try your suggestion shortly.

starbasessd, 2020-02-29:
I have a Pi1 set up with WiFi (Tenda rt2800usb) connected to my SSID: HOME
network
.Unfortunately, I do not have a spare CSI camera available today, but it
successfully connects to one of my IP D-Link DCS-5020L cameras.
Let me know what you need.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-02-29:
My PI reports as:
[root@meye-df2b1254 ~]# cat /proc/cpuinfo
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 697.95
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 000f
Serial          : 00000000df2b1254
Model           : Raspberry Pi Model B Rev 2

ssh or console command line:
cat /proc/cpuinfo <enter>

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

TR-SLimey, 2020-02-29:
So I've tried the newest image and nothing. Same result.
The light on the WiFi dongle blinks implying that there is some activity but for some reason it cannot connect fully. I attach a log from the last boot below. I also quote some parts which I thought were interesting:

```
[   24.117270] ieee80211 phy0: rt2x00lib_request_firmware: Info - Loading firmware file 'rt2870.bin'
[   24.121723] ieee80211 phy0: rt2x00lib_request_firmware: Info - Firmware detected - version: 0.36
[   26.008267] wlan0: authenticate with a0:bd:cd:01:99:f2
[   26.070332] wlan0: send auth to a0:bd:cd:01:99:f2 (try 1/3)
[   26.073484] wlan0: authenticated
[   26.083544] wlan0: associate with a0:bd:cd:01:99:f2 (try 1/3)
[   26.193476] wlan0: associate with a0:bd:cd:01:99:f2 (try 2/3)
[   26.198781] wlan0: RX AssocResp from a0:bd:cd:01:99:f2 (capab=0x1411 status=0 aid=9)
[   26.211559] wlan0: associated
[   40.058802] wlan0: authenticate with a0:bd:cd:01:99:f2
[   40.113845] wlan0: send auth to a0:bd:cd:01:99:f2 (try 1/3)
[   40.115786] wlan0: authenticated
[   40.124077] wlan0: associate with a0:bd:cd:01:99:f2 (try 1/3)
[   40.167293] wlan0: RX AssocResp from a0:bd:cd:01:99:f2 (capab=0x1411 status=0 aid=9)
[   40.180363] wlan0: associated
[   56.038466] wlan0: authenticate with a0:bd:cd:01:99:f2
[   56.093076] wlan0: send auth to a0:bd:cd:01:99:f2 (try 1/3)
[   56.129799] wlan0: send auth to a0:bd:cd:01:99:f2 (try 2/3)
[   56.164003] wlan0: send auth to a0:bd:cd:01:99:f2 (try 3/3)
[   56.168283] wlan0: authenticated
[   56.174103] wlan0: associate with a0:bd:cd:01:99:f2 (try 1/3)
[   56.201547] wlan0: associate with a0:bd:cd:01:99:f2 (try 2/3)
[   56.355093] wlan0: associate with a0:bd:cd:01:99:f2 (try 3/3)
[   56.412602] wlan0: association with a0:bd:cd:01:99:f2 timed out
[   64.557847] wlan0: authenticate with a0:bd:cd:01:99:f2
[   64.614473] wlan0: send auth to a0:bd:cd:01:99:f2 (try 1/3)
[   64.616282] wlan0: authenticated
[   64.624137] wlan0: associate with a0:bd:cd:01:99:f2 (try 1/3)
[   64.628014] wlan0: RX AssocResp from a0:bd:cd:01:99:f2 (capab=0x1411 status=0 aid=9)
[   64.640795] wlan0: associated
[   89.780222] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[  111.045357] wlan0: deauthenticating from a0:bd:cd:01:99:f2 by local choice (Reason: 3=DEAUTH_LEAVING)
```
As for your suggestion with installing Raspbian, I'll get back to you soon.

starbasessd, 2020-02-29:
Can you show the output for lsusb, in raspbian, please?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

TR-SLimey, 2020-02-29:
This is extremely weird but even on Raspbian the Pi refuses to connect even though it worked quite recently. Even when using different USB ports and different dongles. The credentials are 100% correct and the WiFi is in range.

Here's the log from Raspbian which should have some interesting information.

[Raspbian Syslog](https://github.com/ccrisan/motioneyeos/files/4270789/syslog.log)

I will try using a different raspberry pi altogether now.

TR-SLimey, 2020-02-29:
My other Pis refuse to boot altogether. Great.

starbasessd, 2020-02-29:
Same model Pi? If not, need to get image for the new PI...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

TR-SLimey, 2020-02-29:
Yep. Same model according to the writing on the board. I'm really confused.

starbasessd, 2020-02-29:
Can you grab the motionEyeOS 20200203 for Pi?
https://github.com/ccrisan/motioneyeos/releases/download/20200203/motioneyeos-raspberrypi-20200203.img.xz
and install it, please?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-02-29:
Also, can you reboot your wifi router for grins and giggles? To you have a
separate modem & router or all in one?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

TR-SLimey, 2020-02-29:
Okay sooooo, I switched the Raspberry Pi (got one of the other ones working), the WiFi dongle, the power cable and used the image you gave me ([motioneyeos-raspberrypi-20200203.img.xz](https://github.com/ccrisan/motioneyeos/releases/download/20200203/motioneyeos-raspberrypi-20200203.img.xz)) and it connected (without rebooting the router). I switched each and every one of those things (except the pi) out before individually and nothing worked.

And to answer your question, I have both in one though that's probably irrelevant now.

Now to switch each of them back individually and see which is causing the problem (*sighs*)

TR-SLimey, 2020-02-29:
As it turns out, the WiFi dongle was at fault all along. It had worked previously without an issue and now for some reason it's not, regardless of the OS I use. Weird. And the other dongle did not work previously but now all of a sudden it does.

TR-SLimey, 2020-02-29:
Thanks for your help though @starbasessd !

closed 2020-02-29 by TR-SLimey

djfliq1, 2023-10-25:
I'm having this issue as well. My device wifi worked fine until I put Motioneyeos on it. Now it doesn't work and my entire point of using motioneye is to place devices where they are not a wired network connection. I'm not as techy as yall are here, just need to protect my assets and I'm not paying what they want me to pay for a full system, just want my stuff to stop being stolen. I noticed just to boot, you need an internet connection and everything is locked down so I can do things like boot from a USB so I can increase my hdd space on the device. It's a lot. But if yall can help me explain how a known good wifi adapter doesn't work for motioneyeos. I uninstalled motioneye then installed raspian and my wifi works. I re-installed motioneye and wifi don't work. This product should have never been released if wifi is a problem and motioneye is dependent on the internet to boot! My ignorance is what got me here. Need help on getting smart on this like yall are.

starbasessd, 2023-10-25:
@djfliq1 Which Pi, and which wifi? The onboard wifi for all the Pis works out of the box.
If you need to do it wifi only (no ethernet for setup) then you need to add a file to the boot partition called wpa_supplicant.conf, which is documented here:  [wiki for wifi](https://github.com/ccrisan/thingos/wiki/wpa_supplicant.conf)
There are several methods available to boot without internet, but it does require networking either as a client device or as an access point, or you have to specifically tell it to not use any networks.
It has been working for most users for many years. I am available to help, if you give some specifics as what you are using.
At a minimum, which Pi, which version of motionEyeOS you are attempting to use, what WiFi dongle make and model if not the onboard, and whatever else info you may consider important.

starbasessd, 2023-10-26:
@djfliq1 Any further info?

djfliq1, 2023-10-27:
I had chatgpt help me. The issue was the supplicant file, none of my
settings from the radipian installer carried over. Chatgpt walked me
through how to create one so we are good. Sorry to blast you guys but I'm
sure you understand my frustrations.