motioneye-project/motioneyeos #2307: Motioneye poor frame rates / Tiny PC App on Bluestacks good frame rates

bobster316, 2020-02-27:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: Tried all Raspberry Pi 4 versions including dev

### Board Model

I am using the following board/model: Raspberry PI 4  4GB

### Camera

I am using the following type of camera: Wyze Cam v2 via RSTP

My camera model is:  Wyze Cam v2
### Network Connection

My motionEyeOS unit is connected to the network via Ethernet

### Peripherals

Motioneye
https://youtu.be/q6HYTkZaPmU

BlueStacks/Tiny PC
https://youtu.be/BnDZ6oYgj7U

As you can see in the video's the performance of the Raspberry Pi 4 4GB running Motioneyeos is not very good compared to the BlueStacks/Tiny PC.

starbasessd, 2020-02-27:
What are the hardware specs of your Tiny PC?
The CPU on a Pi4 is CPU – Broadcom BCM2711, Quad core Cortex-A72 (ARM v8)
64-bit SoC @ 1.5GHz.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

bobster316, 2020-02-27:
The Tiny PC app is running in BlueStacks on a Windows 10 pc - i7-8700k so i guess there's no comparison. I will never get the frame rates i'm looking for from the raspberry pi. I think i'll give up and go for a standard cctv installation with poe camera's and nvr. 
Thanks