motioneye-project/motioneyeos #2309: Support for monitoring

fredrik-rambris, 2020-02-28:
Is there any support for remotely monitoring cpu, temperature, disk usage etc in motioneyeos?

Something on the lines of collectd, snmpd or even a REST endpoint to get the current stats.

starbasessd, 2020-02-28:
Not built in, but could be added if you wanted
https://github.com/ccrisan/motioneyeos/wiki/Tweaks
https://github.com/ccrisan/motioneyeos/wiki/Action-Buttons
https://github.com/ccrisan/motioneyeos/wiki/FAQ#how-do-i-overlay-output-from-a-command-eg-cpu-temp-on-the-live-image

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

fredrik-rambris, 2020-02-28:
Is there a way to add a script accessible from motioneye (i.e. a URL)? So I
can call http://motion.local/stats   and have it collect some stats and
echo out?

starbasessd, 2020-02-28:
https://github.com/ccrisan/motioneyeos/wiki/Tweaks#running-commands-at-regular-intervals-cron-support

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

HOD42, 2020-03-09:
Could you use prometheus/node_exporter? Works well with grafana.

https://github.com/prometheus/node_exporter/releases/tag/v0.18.1

fredrik-rambris, 2020-03-09:
I might, but I haven't got a prometheus instance running.

closed 2022-01-19 by starbasessd