motioneye-project/motioneyeos #2310: Video Stream is darker than on MotionEye Stream

igloo15, 2020-02-28:
### motionEyeOS Version

I am running motionEyeOS version: motionEyeOS dev20200223

### Board Model

I am using the following board/model: Raspberry PI Zero W

### Camera

I am using the following type of camera: Fast Camera Network and RTSP

My camera model is: Standard NOIR Raspberry Pi Camera

### Network Connection

My motionEyeOS unit is connected to the network via: Wifi

### Peripherals

I am using the following peripherals that I consider relevant to this issue: No Peripherals

### Main Issue

The main issue I am having is that when I log in to the web portal for motioneyeos the video stream I see is considerably brighter and easier to see then when I view the video stream from anywhere else. 

I have tried Blue Iris, Zone Minder, VLC and others. When I view the video streams in both HTTP and RTSP I noticed that they are very dark whether IR Filter is enable or not doesn't make a difference. When I view this supposedly same video stream from the motion eye os web page it looks brighter for some reason. I am trying to figure out why there is a difference.

starbasessd, 2020-02-28:
Do you get the same brighter output when looking at the 8081 stream output
port?
Do you have Fast Network Camera turned on?
Have you changed any other settings?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

igloo15, 2020-02-28:
There is not the worst I have seen but you can clearly see the first image is brighter than the other two. Technically they should be the same but they are not. It is not just VLC it is also blue iris and zoneminder.

This first image is from the motion eye os web page
[image]

This is from rtsp playing to vlc
[image]

This is from http playing to vlc
[image]

This is playing on blueiris same camera no settings changed default
[image]

No settings changes and these pictures taking only seconds apart. I am using Fast Network Camera

igloo15, 2020-02-28:
Whoops look like a cloud or something passed over when I took blue iris it was darker but not that much darker.

[image]

starbasessd, 2020-02-28:
Not having any of the other packages, it may be just the way each processes
the data, converting from one form to another.
If you were interested, I' create a Raspbian Lite setup and install
motionEye on it.
Are you viewing on a PC or Pi?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

igloo15, 2020-02-28:
I am viewing all feeds on the same pc computer. I just find it strange to view the same feed side by side and the one on motionEyeOS be brighter and easier to see.

It doesn't make sense for streams to have different brightness. All streams but the one on motion eye os look the same. So I don't think it is issues with either BlueIris or VLC. The only different one is the MotionEyeOS user interface. Which makes me thing some processing or something is happening to brighten that view.

starbasessd, 2020-02-28:
Do you have Automatic Brightness turned on?
Feed on left is from Streaming URL using VLC, Feed on right is the Web
Gui.using Firefox (concurrent snapshot)
In my case, the VLC seems to be a little sharper and brighter.
[image: image.png]

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

igloo15, 2020-02-28:
Do you mean automatic white balance because yes I have that set to auto

starbasessd, 2020-02-28:
I guess I really can't help, in this case.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-02-28:
One last question:
If you record the video, does it show as VLC/Blue Iris, brightness, or as
Web GUI brightness?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

igloo15, 2020-02-28:
I can't record the video on motion eye since I am using Fast Network Camera

starbasessd, 2020-02-28:
VLC and Blue Iris should be able to record.
You could turn off Fast Network Camera for a test.
Good luck. It would appear to me to be an issue with the other apps, if the
primary source (motionEyeOS) is putting out the best video.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jasaw, 2020-02-28:
Let me chime in here. I implemented the fast network cam RTSP mode for motioneyeos. Viewing RTSP from the camera web interface itself is brighter because the video stream is from camera encoded directly to MJPEG whereas motioneye server decides the rtsp stream and encode to MJPEG.

starbasessd, 2020-02-29:
Interesting. Starting from the same data and source results in 2 different
value levels going to the MJPEG output stream. In the case of a CSI Fast
Netwok Camera, aren't you using the same encoder?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jasaw, 2020-02-29:
In order to output rtsp and MJPEG simultaneously, I had to run h264 and MJPEG concurrently in the GPU for maximum performance. That is why same camera, different brightness.

igloo15, 2020-03-01:
So how do I get them to be the same brightness. As in how do I change the rtsp brightness to match or get closer to the one in MotionEyeOS.

jasaw, 2020-03-01:
Play with the camera settings on the Fast Network Camera web interface. Try increasing the exposure compensation.