motioneye-project/motioneyeos #2311: Problem with ‚pause/start‘ motion detection shell commands

Pedder007, 2020-02-29:
### motionEyeOS Version
I am running motionEye version: 0.41
Motion Version: 4.2.2
OS: Raspbian 10

### Board Model
I am using the following board/model: Raspberry PI 3B

### Camera
I am using the following type of camera: Network via WLAN
My camera model is: Reolink

### Network Connection
My motionEyeOS unit is connected to the network via: Ethernet

### Peripherals
I am using the following peripherals that I consider relevant to this issue: nothing

Since a couple of days I have a problem with the shell commands to switch motion detection on/off. I’m convinced that it formerly worked but now shows up with crucial behaviour.

From my HA Pimatic on Raspi01 I use the standard shell commands to toggle motion detection on/pause on Raspi02:
on command: wget -q "http://192.168.xxx.xxx:xxxx/1/detection/start"
off command: wget -q “http://192.168.xxx.xxx:xxxx/1/detection/pause”

Reading the status by: http://192.168.xxx.xxx:xxxx/1/detection/status
definitely prompts: ‘Camera 1 Detection status PAUSE’, when pause command has been executed.

But motioneye continues to detect motion, or sometimes shortly pauses and then continues after a couple of minutes and sends out mails.
The status stays the whole time on ‚pause‘.

Has someone an idea how to tackle that?

Kind regards
Pedder

starbasessd, 2020-02-29:
Are you using motionEye on Raspbian on Pi3b or motionEyeOS on Pi3b?

What set up instructions did you follow to set this up? (I can try to
replicate the issue, I have the hardware and software available)

Have you tried using 'stop' instead of 'pause'?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Pedder007, 2020-02-29:
Hi @starbasessd,
I directly tested your hint respective using 'stop' in stead of 'pause'.
But it does not work. Checking the status 'http://192.168.xxx.xxx:xxxx/1/detection/status' prompts 'Camera 1 Detection status ACTIVE' and motioneye continuous to detect.
Using 'pause' as I wrote, in minimum deactivates the detection for some minutes.

For the installation I used the official guide here on Github for Raspbian:
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
And also other features work quiet well, incl. the fact that the system really runs stable.
Isn't their a feature (by shell command) that might really toggle the switch in the motioneye GUI? If so, that would really stop the detecion then, and the settings are kept, as I already tested.

starbasessd, 2020-02-29:
Have you tried "Working Schedule"?
Resetting a Pi3B+ for Raspbian & motionEye
Can you post your /etc/motion.conf & motioneye.conf (removing any passwords
and usernames, please.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-02-29:
Sitting here. RPi3B, Raspbian Lite, motionEye latest through the install
shown.
4 cameras set up (3 D-Link, 1 MotionEyeOS)
Set up all 4 cameras, the motionEyeOS cam as #1
enabled /etc/motioneye/motion.conf web settings
webcontrol_port 7999
setup_mode off
webcontrol_parms 2
webcontrol_localhost off
webcontrol_interface 1

rebooted
WebGUI to :9765, displaying all 4 cameras.
Set camera1 to motion detect:
[image: image.png]
Connected to http://<IP>:7999/1/detection/status
Status ACTIVE
Connect to :7999/1/detection/pause
Status PAUSE
Wait 5 minutes
make change in another camera, click Apply
Connect to http://<IP>:7999/1/detection/status
Status ACTIVE
Connected to http://<IP>:7999/1/detection/pause
Status PAUSE
check in 5 minutes
Connected to http://<IP>:7999/1/detection/status
Status PAUSE
make similar change in another camera
Connected to http://<IP>:7999/1/detection/status
Status ACTIVE
Connected to http://<IP>:7999/1/detection/pause
Status PAUSE
wait 30 minutes, only viewing cameras
Connected to http://<IP>:7999/1/detection/pause
Status PAUSE
Working as expected.
During the active times, records activity
During PAUSE, no recording, no notifications.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-02-29:
Further testing:
If I create a
sudo crontab:-e
-----

*/5 * * * * sudo rm -f /home/<user>/status && wget
http://localhost:7999/1/detection/status -O /home/<user>/status
-----
it should delete and create the status file every 5 minutes on the Hub
(change <localhost> to IP for another Hub)
You could then get the end of the text line and force start or pause as
needed.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Pedder007, 2020-02-29:
Wow, a lot of work you did that quick, thank you! .. and sorry I had some other stuff in the meantime ...

However, I just attached my 2nd cam to make testing a bit easier. And again cruical, 'it' does not show the same 'wrong' behavior. The configuration is exact the same as the other cam.

Sending 'pause' keeps motion detection reproducable off for >15minutes. I did it now 3 times plus some shorter timeframes. Additionally I also sent the 'pause' command two and three times, to prevent a maybe cruical behavior if my HA sends it twice or more often.
'Working schedule' I never used, as the cam in general runs 24hrs. Only motion detection 'pause' is needed for dedicated situations, not to produce hundreds of needless pictures.
What do you mean by 'resetting'? I for sure rebooted the motioneye PI3B in the last days in this context here.
With respect to your second post: I couldn't find such a status file on the RP3B!?

I will let the 2nd cam run now 'paused' overnight, to watch out for maybe 'long term' problems....

I will test again tomorrow with the outside-cam and maybe completely delete it from motioneye and configure it new from scratch. 

My motion.conf:
=====================
# @enabled on
# @normal_password 
# @admin_username 
# @admin_password 
# @normal_username 

webcontrol_interface 1
webcontrol_port 7999
setup_mode off
webcontrol_parms 2
webcontrol_localhost off

camera camera-1.conf
camera camera-2.conf
=====================

and the motioneye.conf:
=====================
 path to the configuration directory (must be writable by motionEye)
conf_path /etc/motioneye

# path to the directory where pid files go (must be writable by motionEye)
run_path /var/run

# path to the directory where log files go (must be writable by motionEye)
log_path /var/log

# default output path for media files (must be writable by motionEye)
media_path /var/lib/motioneye

# the log level (use quiet, error, warning, info or debug)
log_level info

# the IP address to listen on
# (0.0.0.0 for all interfaces, 127.0.0.1 for localhost)
listen 0.0.0.0

# the TCP port to listen on
port 88
motion_check_interval 20

# path to the motion binary to use (automatically detected if commented)
#motion_binary /usr/bin/motion

# whether motion HTTP control interface listens on
# localhost or on all interfaces
motion_control_localhost true

# the TCP port that motion HTTP control interface listens on
motion_control_port 7999

# interval in seconds at which motionEye checks if motion is running
motion_check_interval 10

# whether to restart the motion daemon when an error occurs while communicating with it
motion_restart_on_errors false

# interval in seconds at which motionEye checks the SMB mounts
mount_check_interval 300

# interval in seconds at which the janitor is called
# to remove old pictures and movies
cleanup_interval 43200

# timeout in seconds to wait for response from a remote motionEye server
remote_request_timeout 10

# timeout in seconds to wait for mjpg data from the motion daemon
mjpg_client_timeout 10

# timeout in seconds after which an idle mjpg client is removed
# (set to 0 to disable)
mjpg_client_idle_timeout 10

# enable SMB shares (requires motionEye to run as root) 
smb_shares false

# the directory where the SMB mount points will be created
smb_mount_root /media

# path to the wpa_supplicant.conf file
# (enable this to configure wifi settings from the UI)
#wpa_supplicant_conf /etc/wpa_supplicant.conf

# path to the localtime file
# (enable this to configure the system time zone from the UI)
#local_time_file /etc/localtime

# enables shutdown and rebooting after changing system settings
# (such as wifi settings or time zone)
enable_reboot false

# timeout in seconds to use when talking to the SMTP server
smtp_timeout 60

# timeout in seconds to wait for media files list
list_media_timeout 120

# timeout in seconds to wait for media files list, when sending emails
list_media_timeout_email 10

# timeout in seconds to wait for zip file creation
zip_timeout 500

# timeout in seconds to wait for timelapse creation
timelapse_timeout 500

# enable adding and removing cameras from UI
add_remove_cameras true

# enables HTTP basic authentication scheme (in addition to, not instead of the signature mechanism)
http_basic_auth false

# overrides the hostname (useful if motionEye runs behind a reverse proxy)
# server_name motionEye
======================

starbasessd, 2020-03-01:
Both motion.conf and motioneye.conf are in the /etc/motioneye folder,
correct?
If I don't touch the WebGUI for motionEye (port 8765, settings) If I pause
it, it stays paused. If I touch anything in settings in the WebGUI for any
camera, it resets the status back to ACTIVE.
In the   sudo crontab -e    command, the status file appears in the
/home/<user>/ folder  << Change <user> to whatever user you normally log on
as.
It shows me the last status (checked every 5 minutes) if I want to grep for
PAUSE or ACTIVE and do something about it.
Because it sits on mine at whatever status I select until I tell it to
'start' again, or make a change, not a random time, I don't think it's
motioneye or motion.
Let me know what result you get in the AM.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-03-01:
12 hours result: Still Paused.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Pedder007, 2020-03-01:
Hey @starbasessd,
the two files are exactly in your mentiones folders.
I know the effect by touching the WEBGUI, that's not the problem.
I think I got your point now respective the status. I cover this manually by calling the http-link I already mentioned above: 'http://192.168.xxx.xxx:xxxx/1/detection/status'
Using it, always shows me the correct status. So ACTIVE if before called 'start' or PAUSE if befor called 'pause'. And that was exactly why a had no clue how to tackle the problem.

But now with respective to my additionally configured cam:
It is now running since last night in 'pause' modus and does what it should: no detection and no notification.
But the best comes now:
Since then, respectively since this morning, when time by time someone is passing our front-door, it works also perfectly!?!?!? No detection as long as it is in 'pause' modus - isn't that wondering?
I have no clue how that came now. I will now disable again my second cam to check if that might be a driver to change the behavior of the problem cam.
If so, as I wrote, I will completely delete the problem cam from motioneye and configure ist new.

Thanks for your effort again for the moment!!!
Pedder

starbasessd, 2020-03-01:
Let me know.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Pedder007, 2020-03-01:
OK, the 5 tests today, after switching off the second cam, have been successful.
If cam No1 has been set to 'pause', no motion has been detectes and so no pictures have been shot.
We will proceed tomorrow during a normal working day, when the 'door-frequence' is definitely higher. I will set the pause directly to 1 hour, bevor we individually leave the house.
Let's see ....

starbasessd, 2020-03-01:
Good deal.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Pedder007, 2020-03-03:
Hi again, after further testing the start/pause functionality works without any problems.
Whatever didn‘t work, I have no clue, I didn‘t change anything, besides temporarily having the second cam connected.
It is unsatisfying not to know what the problem was about, but I think I will close the issue.
What do you mean?

starbasessd, 2020-03-03:
Just that I meant you were going to continue testing your specific setup.
I know it's frustrating when you can't see a cause for a glitch, or that
something 'fixes itself' while poking around.
If it reoccurs feel free to post here again.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Pedder007, 2020-03-04:
OK, I will close here now and oberserve further on, ...
@starbasessd thanks again for your effort 👍

closed 2020-03-04 by Pedder007