motioneye-project/motioneyeos #2312: problems with motion detection

PeterEdelmann, 2020-02-29:
Hello,
I have the followin problem 
I motion is detected the stream freezes.
I found the folloing message i motioeye.lo
2020-02-29 17:54:28: [motioneye]    ERROR: failed to create movie preview for /data/media/sda3/2020-02-29/15-36-31.mp4: Command '[u'ffmpeg', u'-i', u'/data/media/sda3/2020-02-29/15-36-31.mp4', u'-f', u'mjpeg', u'-vframes', u'1', u'-ss', u'20', u'-y', u'/data/media/sda3/2020-02-29/15-36-31.mp4.thumb']' returned non-zero exit status 1
After reboot it works till the next Motion detected.
I tried installing the System new on a Raspberry 3. It was not the solution.
Could it be a Problem with rasperry itself?
Help would be appriciatid
PS: The System worked since 1 year without Problems. I have it since one week, I didnt Change anything

starbasessd, 2020-02-29:
Which version of motionEyeOS?
External USB drive?
Does the folder tree exist: /data/media/sda3?
In Console or ssh:  ls -la /data/media/sda3
Is it full?
In console or ssh:   df -h

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

PeterEdelmann, 2020-02-29:
Version is 

motionEye Version

0.41

Motion Version

4.2.2

OS Version

motionEyeOS 20190911

 

Ist an external usb (but I tried it with the sdcard of raspberry too, the same)

Please look at the attached png

starbasessd, 2020-02-29:
Don't see any attachments.

PeterEdelmann, 2020-02-29:
Welcome to meye-b28ee4a2!

admin@192.168.178.35's password:

[root@meye-b28ee4a2 ~]# ls -la /data/media/sda3

total 64

drwxr-xr-x   13 root     root          4096 Feb 29 19:27 ./

drwxr-xr-x    5 root     root          4096 Feb 29 12:24 ../

-rw-rw-rw-    1 root     root             0 Feb 29 19:56 .keep

drwxr-xr-x    3 root     root          4096 Jan  1  1970 .overlay-log/

drwxr-xr-x    3 root     root          4096 Jan  1  1970 .overlay-usr/

drwxr-xr-x    3 root     root          4096 Jan  1  1970 .overlay-varlib/

-rw-r--r--    1 root     root             0 Jan  1  1970 .post-upgrade-net-scheduled

drwxr-xr-x    2 root     root          4096 Feb 29 19:34 2020-02-29/

drwxr-xr-x    4 root     root          4096 Feb 29 11:59 etc/

drwxr-xr-x    3 root     root          4096 Jan  1  1970 log/

drwx------    2 root     root         16384 Jan  1  1970 lost+found/

drwxr-xr-x    3 root     root          4096 Feb 29 11:49 media/

drwxr-xr-x    2 root     root          4096 Sep 10 23:44 output/

drwxr-xr-x    2 root     root          4096 Jan  1  1970 usr/

drwxr-xr-x    4 root     root          4096 Feb 29 11:48 varlib/

[root@meye-b28ee4a2 ~]# df -h

Filesystem                Size      Used Available Use% Mounted on

/dev/root               213.0M    197.1M    903.0K 100% /

devtmpfs                369.3M         0    369.3M   0% /dev

tmpfs                   373.8M     16.0K    373.8M   0% /dev/shm

tmpfs                   373.8M      2.3M    371.5M   1% /tmp

/dev/mmcblk0p1           29.9M     14.1M     15.8M  47% /boot

/dev/mmcblk0p3           13.2G     47.2M     12.4G   0% /data

/dev/mmcblk0p3           13.2G     47.2M     12.4G   0% /home/ftp/sdcard

/dev/mmcblk0p3           13.2G     47.2M     12.4G   0% /home/ftp/storage

/dev/sda2               213.0M    186.8M     11.2M  94% /data/media/sda2

/dev/sda2               213.0M    186.8M     11.2M  94% /home/ftp/storage/sda2

/dev/sda3                28.2G     81.2M     26.7G   0% /data/media/sda3

/dev/sda3                28.2G     81.2M     26.7G   0% /home/ftp/storage/sda3

/dev/sda1                29.9M      6.0K     29.9M   0% /data/media/sda1

/dev/sda1                29.9M      6.0K     29.9M   0% /home/ftp/storage/sda1

[root@meye-b28ee4a2 ~]#

starbasessd, 2020-02-29:
What changed a week ago?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

PeterEdelmann, 2020-02-29:
As far as I know nothing. I restarted my router as repeater a few times. It's in a mesh system connected 5gh with a vm router 7590. Raspberry is there Lan connected. Von meinem Huawei-Tablet gesendet-------- Originalnachricht --------Betreff: Re: [ccrisan/motioneyeos] problems with motion detection (#2312)Von: starbasessd An: ccrisan/motioneyeos Cc: PeterEdelmann ,Author What changed a week ago?




-- 
Thanks

Kevin Shumaker


Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?


[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/ccrisan/motioneyeos/issues/2312?email_source=notifications\u0026email_token=AOWGOFKKUQNHZKY5M53A63DRFFUY3A5CNFSM4K63L6Z2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOENMETYY#issuecomment-592988643",
"url": "https://github.com/ccrisan/motioneyeos/issues/2312?email_source=notifications\u0026email_token=AOWGOFKKUQNHZKY5M53A63DRFFUY3A5CNFSM4K63L6Z2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOENMETYY#issuecomment-592988643",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

starbasessd, 2020-02-29:
Something is preventing you from writing to the /data/media/sd3 folder
tree.
Did a search for " Command '[u'ffmpeg', u'-i', u'/data/media/sd" and see
several for motionEye.
Out of curiosity, what setting are you using for Movies, Movie Format?
Can you try H.264 (mp4) or MPEG4 (avi) and see if that helps?

PeterEdelmann, 2020-02-29:
I had movie format h.264/OMX. Now I use h.264 (.mp4) Tell you tomorrow, if it helped.

Thanks so far

PeterEdelmann, 2020-03-01:
Hello, It worked again without problems in the moment with h. 264. I do not know what I have done. Thanks a lot for your help. PeterVon meinem Huawei-Tablet gesendet-------- Originalnachricht --------Betreff: Re: [ccrisan/motioneyeos] problems with motion detection (#2312)Von: starbasessd An: ccrisan/motioneyeos Cc: PeterEdelmann ,Author Something is preventing you from writing to the /data/media/sd3 folder
tree.
Did a search for " Command '[u'ffmpeg', u'-i', u'/data/media/sd" and see
several for motionEye.
Out of curiosity, what setting are you using for Movies, Movie Format?
Can you try H.264 (mp4) or MPEG4 (avi) and see if that helps?

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/ccrisan/motioneyeos/issues/2312?email_source=notifications\u0026email_token=AOWGOFIVOZWHJFB4EVUN7STRFFX4ZA5CNFSM4K63L6Z2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOENMFI3Y#issuecomment-592991343",
"url": "https://github.com/ccrisan/motioneyeos/issues/2312?email_source=notifications\u0026email_token=AOWGOFIVOZWHJFB4EVUN7STRFFX4ZA5CNFSM4K63L6Z2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOENMFI3Y#issuecomment-592991343",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

starbasessd, 2020-03-01:
You didn't do anything. Apparently there is/was a known issue with the OMX
conversion that happened a while back. If you select another format, it
bypasses the issue.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2022-01-19 by starbasessd