motioneye-project/motioneyeos #2313: RPi4 64 bit support

roger-, 2020-02-29:
Seems like there could be a significant performance gain by switching: https://medium.com/@matteocroce/why-you-should-run-a-64-bit-os-on-your-raspberry-pi4-bd5290d48947

ccrisan, 2020-03-01:
Interesting benchmarks! Just like the author of the article, I lived with the impression that performance gains would be minimal. I'll have to look into this and see what we'd have to lose by switching to 64 bit.

One thing I'm concerned about is the kernel. We rely on the exact same source tree and configuration used by Raspbian. If we drift apart, we might get into unnecessary trouble.

roger-, 2020-03-01:
There’s a 64 bit kernel here FYI: https://www.raspberrypi.org/forums/viewtopic.php?t=250730