motioneye-project/motioneyeos #2314: Raspberry Pi 3A+ not supported!

buzzy, 2020-03-02:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190911.

### Board Model

I am using the following board/model: Raspberry PI 3A+.

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 
As the Raspberry Pi 3A+ does NOT have a ethernet port, the whole system halts at boot, complaining about no wired network and then reboots. Why is the 3A+ listed as supported in the list when it obviously is not supported?

starbasessd, 2020-03-02:
With the 3A+ you'll need to
pre-load the wpa_supplicant.conf file in /boot with your network settings
or
set OS_NETWORKLESS="true" in os.conf and put it into /boot (disable
networking temporarily)
BEFORE starting motionEyeOS the first time.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

buzzy, 2020-03-02:
I flashed the latest stable to an SD-card. Then mounted the second partition and created a file called "os.conf" in the /boot directory with the contents:

set OS_NETWORKLESS="true"

That causes a faulty boot that just shows:

can't open /dev/ttylogin: No such file or directory.

I also tried it without the word "set". Then it boots, but still complains about no network and reboots.

starbasessd, 2020-03-02:
How do you image the SDCard? From PC or Linux?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

buzzy, 2020-03-02:
Not sure I understand what you mean? PC is a computer and Linux is an operating system. Are you asking me if I am using Windows, MacOS or Linux? I am using Linux on a PC.

starbasessd, 2020-03-02:
OK. when you image the SDCard, mount the /boot partition

If you want to use WiFi:
create a file called wpa_supplicant.conf in the /boot partition
put the following contents in it (without the "-----" lines, and put in
your SSID and PASSWORD where I put SSID and PASSWORD)
-----
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
    ssid="SSID"
    psk="PASSWORD"
key_mgmt=WPA-PSK
}
-----
eject, insert into Pi3A+ and boot. It should connect to your WiFi.

If you want stand alone without network connection:
I have attached a zip file with the needed config files and instructions to
make it a Wireless Access Point so you can get to the WebGUI and make your
settings available to a phone, tablet or laptop.

I'm here to help.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

buzzy, 2020-03-02:
Ok, I will look at it again. However, I am pretty sure there are only 2 partitions. One FAT-partition that will most likely be the boot-partition and then another one for the linux root. You told me to put the file in the /boot which would mean the "boot" directory on the second partition.

Are you now saying that I should put it on the first partition? It might be that the first partition get mounted in "/boot" after a normal boot, but when mounting the SD-card on another computer to add the file you are mentioning, that mount has of course not happened. I will try it and let you know. Thanks!

buzzy, 2020-03-02:
I tried putting the wpa_supplicant.conf in the root of the first partition. The system seems to have picked it up, but is unable to connect to the Access Point. By now, I think something is wrong with my board, as I have been unable to connect to Access Points from several different distributions, including Rasbian and Ubuntu Server. I can list Access Points without any issues, but wpa_supplicant always returns wlan0: CTRL-EVENT-ASSOC-REJECT

starbasessd, 2020-03-02:
That is correct. The partitions are labeled boot and root. The FAT is boot,
and will eventually be mounted as /boot folder.
Root is EXT4. (You may also want to look at mounting overlay file systems,
too, as they are used extensively in motionEyeOS)
The scripting in motionEyeOS (and Raspbian) is based on putting your
pre-booting scripts and config files in the boot partition prior to first
startup.
That is one reason the first partition is FAT, so Windows can also access
it, as there isn't any reliable support for EXT4 partitions in Windows.
Another issue for you will be most of the system is mounted R/O (read only)
and cannot be modified easily (by design)
Let me know if you need further assistance.
You may want to upgrade to 20200203, it's the latest sable. dev20200223 is
latest 'bleeding edge' dev version.
https://github.com/ccrisan/motioneyeos/releases

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-03-02:
Interesting.
You may be correct in that your WiFi has issues. I can recommend testing
with a wifi dongle, too.
A quick google search points to a defective wifi adapter/driver, or a bad
character in the SSID.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

buzzy, 2020-03-02:
Yes, I have tried that latest version as well. Same result. I think something might be wrong with the built-in WiFi-card on my board. I never got it to connect to an Access Point on any distribution, including the official Rasbian. Even tried with the desktop version using the GUI. The even more weird part is that external WiFi dongles have the issue too. Dongles that works perfectly on my old Raspberry Pi 2.

starbasessd, 2020-03-02:
Does your Pi2 still work with the dongle?
Have you done any upgrades to your router?
What's changed between the 2 and 3A?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-03-02:
Oh, Have you rebooted your WiFi router? Say that as a fix on a RPi Forum @
raspberry pi foundation.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

buzzy, 2020-03-02:
My router does not have anything to do with it. I have tried it with several Access Points, but I mainly use my phone as a hotspot/AP when configuring things like this. Yes, the dongle still works on my old RP2. I have 4 different dongles with different chipsets and they all work on the RP2, but not on the RP3. I don't know all the changes between the models, but they probably upgraded the CPU etc. the RP3 has built-in WiFi card. The RP2 does not.

starbasessd, 2020-03-02:
Since you didn't fully fill out the original information at the beginning,
I have to ask questions, I'm sorry.
Rebooting hardware is one of the first diagnostics we (Corporate IT Help
Desk, my profession).do, to start clean.
Yes, RPi has upgraded just about everything from the Pi2 to the Pi3A+, but
Raspbian still supports all of them from the original through the Pi4Bs.
It automatically shifts kernels, firmware, etc, as needed.
So, at this point, I would have to agree with your assessment that it's a
defective PI3A+.
If anyone else wants to jump in, feel free.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

buzzy, 2020-03-02:
Yeah, I agree that restarting usually helps with people that does not know anything about IT. BUT, even IF restarting would solve an issue, you still don't know the cause and therefor, in my opinion, the problem has not really been solved as it will probably happen again. But in any case, I appreciate that you took your time to try to help me. At this point, I will just try to return the board.

TechNovice79, 2020-03-12:
@buzzy I'm not a Linux expert, but I'd like to help troubleshoot. Restarting is usually my second step (the first is searching the internet). Where/when did you purchase the 3A+? You mentioned that you tried Raspbian. What version(s) of Raspbian have you tried? What error messages (if any) did you get when you tried to use the GUI to connect? Have you tried accessing a different WiFi AP? Have you tried a different microSD card? When you are in Raspbian, what do you get when you type `lsusb` in the terminal/command line? What about `ifconfig -a`? Are you using any other peripherals/hats with the Raspberry Pi? I have read that some devices can cause WiFi problems with the Raspberry Pi.

When you get the `wlan0: CTRL-EVENT-ASSOC-REJECT` error, does it give you a status_code after? Many of those errors I'm seeing online are when someone is trying to connect to their phone as an AP. Have you tried connecting to an AP that is not a phone?

I have a 3A+ and it works with MotionEyeOS 20190911.

buzzy, 2020-03-12:
Thanks for helping!
1. I have restarted several times. Both the board and all Access Points.
2. I bought my Raspberry from a seller on AliExpress.
3. I tried the latest Raspbian Buster and the special Raspberry version of Ubuntu Server.
4. GUI shows a list of APs but gives error when trying to connect to one (Failed to connect to Network)
5. I have tried 3 different routers and 2 different phones as Access Points. With and without WPA.
6. Yes, 2 different SD card. Worth noting is that the SD card works fine in another Raspberry board.
7. lsusb and ifconfig -a shows the wlan chip just fine and "iwlist wlan0 scan" shows all APs.
8. I do not have anything else connected to the Raspberry except keyboard and monitor.
9. I get "error 16" after the CTRL-EVENT-ASSOC-REJECT message.

TechNovice79, 2020-03-13:
Have you tried using Raspbian Stretch rather than Buster?

buzzy, 2020-03-14:
I just tried Stretch. Same problem. I have looked over the board and everything looks fine. It looks like my other working boards as far as I can tell.

Daloa1, 2025-11-21:
Hey everyone, did you get it to work by any chance... I had a spare 3 A+ here and also tried setting up MotionEyeOS but unfortunately I just couldnt get it to work...
SD Card is finde since other OS would load on the PI, also my hdmi connection works but as soon as I try to get MotionEyeOS on it with WIN32Imager or PI Imager i believe it goes in a Loop Boot since the green LED is starting to flicker faster and faster until it stops.
I believe my Installation is correct since I watched pretty much every video on Youtube on how to get it to work.
Anyone HELP please...