motioneye-project/motioneyeos #2315: No, you don't. motionEyeOS is not Debian or a general purpose distro: if you configure it to work with WiFi, it will reboot indefinitely until it gets WiFi.

mickrick1, 2020-03-03:
No, you don't. motionEyeOS is not Debian or a general purpose distro: if you configure it to work with WiFi, it will reboot indefinitely until it gets WiFi.

_Originally posted by @ccrisan in https://github.com/ccrisan/motioneyeos/issues/42#issuecomment-173836162_

I am a bit stuck and hoping you can help.
I got Motioneye working with a Pi camera and everything was rosy (apart from trying to find suitable motion detection settings). I decided to tweak the network settings to fix the IP address. As I clicked 'apply', I saw, to my horror, that I had put 198.162.1.107 instead of 192.168.0.170 as all my devices connect to a Virgin Hub 3.0 on the 198.162.0.xxx mask.
I could not find a way to correct this as the motioneye set-up was not unobtainable from my browser any longer. 
I have reformatted and recopied the motioneye image three time to the sd card but, each time on boot, I can see the motioneye set-up trying and succeeding to connect wlan0 to 198.162.1.107 which is unviewable to me.
Is this setting stored on the Pi itself somehow as reformatting the card must delete all the info there? If so, how do I change or delete it using the motioneyeOS?
I'm a newbie to pi's and motioneye so am following published guides a little slavishly and could not find anything to help me so am turning to you to, hopefully, get me out of this jam. Thanks in advance.

starbasessd, 2020-03-03:
If you are using the latest release, you can place a new wpa_supplicant.conf file in the /boot partition from the machine you used to image it, correcting the data. Helpful on a PiZeroW or Pi3A+.
On any other Pi, you can connect to ethernet to get it to boot,  to correct the wireless settings.
In either case, you should be able to re-image the SDCard and copy the wpa_supplicant.conf to the /boot partition.

mickrick1, 2020-03-03:
Thanks, starbasessd. 
I have place the wpa_supplicant.conf file in the boot directory everytime I reflashed the sd card but that file only contains information about the SSID and password - it does not say anything about the IP address to use so, I assume, it's allowing my router DHCP to pick an available IP address. Do you know if there's an option to put some other information into this file to force it to use a particular IP address - in my case, 192.168.0.107?

starbasessd, 2020-03-03:
Lots of tweaks on the right at:
https://github.com/ccrisan/thingos/wiki/
You may be looking for:
https://github.com/ccrisan/thingos/wiki/static_ip.conf

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-03-03:
BTW, If I need a static IP address, I do it on the router as a reserved
address. That way, If I have a router failure, and have to replace it with
a new one, I get a dhcp address that works, and can get back online
faster...  YMMV

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mickrick1, 2020-03-03:
Thanks again for taking the trouble to reply.
My router has 192.168.0.107 reserved for the motioneye set-up in my Zero W. Unfortunately, the motioneye boot somehow assigns wlan0 to 192.168.1.107 - that's what's confusing me. It's as if it's burnt into the pi's memory even though the only memory of the board is provided by the sd card and that's been reformatted so can't carry any info about the IP address. I will take a look at the links you have provided - thanks again.

starbasessd, 2020-03-03:
Do you have another SDCard? Format it with Raspbian, put in your
wpa_supplicant.conf file, and see if it pulls the correct dhcp address.
If it does, then format it with motionEyeOS, use the same
wpa_supplicant.conf file, and see what happens.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mickrick1, 2020-03-03:
I do have another card formatted with Raspian and when I put it into this Zero W it started up and connected to my wifi at 192.168.0.71 proving, I guess, that the IP address is set on the card not the Pi. Unfortunately, when I replaced the motioneyeOS card, it went back to setting wlan0 to 192.168.1.107 which remains invisible. 
I can log into motioneye using the default username - do you know if I can change the IP address there? If I can, I can view the camera from a browser and remove the fixed IP address that I incorrectly set.

starbasessd, 2020-03-03:
OK, take the card that you put raspbian on. Put motionEyeOS 20200203 on it
with the working wpa_supplicant.conf file. Not the apparently broken card.
What IP address does it show?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mickrick1, 2020-03-03:
I'll get a new blank card and let you know - thanks for your advice so far.

bbillp, 2020-03-04:
The device assignment is in the router , access the router table for the Pi MAC address and clear or change it.

mickrick1, 2020-03-04:
Thanks Bill. I checked the table under the DHCP tab in my router and that has 192.168.0.107 reserved for the motioneye set-up. I added the MAC to the reserved list to be sure.
The device does not show up when it's on as connected in my main router nor in my repeater bridge. 
I think I'll make another wpa_supplicant.conf file and go and try it on another wireless network while I wait for my new sd cards.
Attached: Images of Pi startup messages and entry on reserved IP addresses table
[image]
[image]

starbasessd, 2020-03-04:
Are you using motionEyeOS 20200203? If not, PLEASE try it.
You do not need the os.conf file, just the wpa_supplicant.conf file. It
looks like your Static IP reservation is set up correctly (as long as the
MAC is correct)
You can use either SDCard you have.
What program did you use to create your wpa_supplicant.conf file?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mickrick1, 2020-03-04:
I used Notepad++ with the EOL option set to UNIX.
Am downloading 20200203 issue now and will reflash.
Thanks again.

starbasessd, 2020-03-04:
Excellent. Let me know how it goes.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mickrick1, 2020-03-04:
Huzzah! The new image file did the job! Thanks for all your help. You really saved the day.

Now I just have to go back to sorting out the motion detection settings - I seem to get either nothing or 10 pictures a minute. Just want to find a happy medium that will capture moving persons etc. and not every twitch of a leaf. 

[image]

starbasessd, 2020-03-04:
Excellent!
I have my motion detection Frame Change Threshold set to .1% to get my
'happy medium', you can try turning on and setting up a Mask to only watch
areas of importance, rather than trees.
When I first set up cameras, I turn on Show Frame Changes, too, to see what
is triggering.

[image: image.png]

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2022-01-19 by starbasessd