motioneye-project/motioneyeos #2316: Can't open in web browser. Please help!

danialaris, 2020-03-04:
I have changed the ip address in motioneye from original to a static ip address. Now I can't open in web browser. How do i fix it?

starbasessd, 2020-03-04:
What is your setup?
What hardware? Which SBC? Pi 1,2,3,3A+, 3B, 3B+, 4B, Zero, oDroid, other?
Which version of motionEye or motionEyeOS? (try to use 20200203)
How are you connecting it to your network? WiFi or Ethernet?
What are you using to connect to it? Windows PC, MAC, or Linux?
Which Browser?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

danialaris, 2020-03-04:
> What is your setup? What hardware? Which SBC? Pi 1,2,3,3A+, 3B, 3B+, 4B, Zero, oDroid, other? Which version of motionEye or motionEyeOS? (try to use 20200203) How are you connecting it to your network? WiFi or Ethernet? What are you using to connect to it? Windows PC, MAC, or Linux? Which Browser?
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

I'm using a raspberry pi 3B+, using the latest version of motioneyeos. connecting using wifi. using Windows. I can't seem to open using the static IP address that i changed from the original. It was working before, but after i changed the IP address I can't open it. How can i fix it? Can i fix it through root@meye command?

starbasessd, 2020-03-04:
Can you ping the ip address you set?
Can you connect a monitor to the Pi, and see if the IP address you set is
the same as what shows on the monitor?
Need to see what IP address it is using.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

danialaris, 2020-03-04:
> Can you ping the ip address you set? Can you connect a monitor to the Pi, and see if the IP address you set is the same as what shows on the monitor? Need to see what IP address it is using.
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

It uses the IP address that I set. but i can't open it by typing the IP address on the web. I changed the IP address on the motioneyeos web page when i was still using the original IP address. After I changed it I can't open it.

starbasessd, 2020-03-04:
Again, short of re-imaging, and starting over, can you put a monitor on the
Pi and see what address it is using? It displays it near the bottom of the
boot up text. You may have set it wrong.
Or you may have set it to an address already being used by another device.
Another user I just helped had set it to 192.168.1.x when he needed to set
it to 192.168.0.x, a simple typo, but hard to fix without re-imaging.
If you use 20200203 I can shoe you a config file to set a static IP, or you
can set the static IP in your router.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

danialaris, 2020-03-05:
Yes I can put a monitor on the Pi, the IP address is the IP address that I changed. Maybe I have set it on an IP address that is used, but when I scan using IP scanner, there is no device using the IP address that i set for the MotionEyeOS. I'm using 20190911. Should I re-imaging? Do I need to reformat the SD card? or just use Etcher and re-imaging?

starbasessd, 2020-03-05:
Reimage with 20200203. Just use Etcher.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?