motioneye-project/motioneyeos #2317: Problems with Motiondetection/Recording

Wurstoma, 2020-03-04:
Hello,

i installed motioneyeos 2 weeks ago. It works really good, but i have two little problems.
My recording mode is recording continuously with maximum movie length 600 seconds.
So i have seperate 10 minutes movies over the whole day.
My idea was to turn the motion detection with email notifications on, so i can check it out when a motion was detected. The email notification works.
But my problem is, i am additionally receiving an email every 10 minutes when a new video starts to record. Now i receive a lot of emails every day and its not possible to find out which one of them is a real motion detection email.
I want to maintain the continuous recording but without receiving an email every 10 minutes.
Is it possible?

The second problem is:
I want to record my video continuously but make pictures motion triggered.
The selection for recording mode changes for video and pictures.
Is it possible to select different recording modes for video and picture?

greetings Patrick

### motionEyeOS Version

motionEyeOS 20190911

### Board Model

Raspberry PI 4 Modell B; 4GB

### Camera

Two Cameras: Foscam FI9900P

### Network Connection

Ethernet

starbasessd, 2020-03-04:
1st and second questions are related, and a design limitation, @ccrisan and
others may be able to confirm.
If you select Continuous Movies, you cannot select Motion Detection for
Pictures. If you select Motion Detection for Pictures, it will
automatically select Motion Detection for Movies. I believe it has to do
with the way motion (the underlying program that does the detection) works.
I do not believe it can be set up the way you want. If you set one Pi to
continuous record, and a second Pi to do only motion detect and email, as a
work-around, you should be able to get what you are looking for.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sts-ryan-holton, 2020-03-05:
Maybe linked to my unanswered issue from December? https://github.com/ccrisan/motioneyeos/issues/2233