motioneye-project/motioneyeos #2318: Camera stops streaming constantly requiring power cycle to work

kalex1114, 2020-03-05:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.
yes
I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.
yes
### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).
motionEyeOS dev20200223## Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).
Raspberry Pi 3 Model B Plus Rev 1.3
### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).
V4L2 Camera
### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).
Ethernet
### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)
none
### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.l
[motion.log](https://github.com/ccrisan/motioneyeos/files/4294895/motion.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/4294896/messages.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/4294898/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/4294899/dmesg.log)
[motioneye (1).log](https://github.com/ccrisan/motioneyeos/files/4294900/motioneye.1.log)

Been happening for a while.  It works fine and I'm able to get a feed in my blue iris v4.  After a while I get No signal in blue iris.  Logging into motioneye web gui i get a spinning circle on the camera.  rebooting doesn't fix the issue but if i remove power and then power it on it starts working again.

Any ideas?

kalex1114, 2020-03-08:
anyone?

starbasessd, 2020-03-08:
I don't have a Blue Iris machine, but can reproduce everything else.
I have a Generic USB (TIGA) camera.
I use 20200203 (Non Dev) for Pi3B+
I have GigBit Ethernet.
I Use both Network Watch and HTTP watch.
I cannot reproduce your issue even with the Watch Functionality turned off.
For testing purposes, can you install the 2020-02-05 version of Raspbian
and install motionEye with the following setup instructions to see if it
crashes?
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

kalex1114, 2020-03-09:
Camera stops responding in motioneyes itself. When I access the webui in motioneyes I just get spinning circle and no video. I’ll try your link

starbasessd, 2020-03-09:
Let me know, please.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

kalex1114, 2020-03-09:
I'm not seeing 2020-02-05 version.  i do see
Raspbian Buster Lite
2020-02-13

would this work?

starbasessd, 2020-03-09:
Yeah, sorry. Too many systems, too many naming conventions... <sigh>

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

kalex1114, 2020-03-11:
Looks like I'm getting same issue with this install as well.  Camera stops responding but webui still works fine

starbasessd, 2020-03-11:
I am confused. (Easy at my age)
You say you have a V4l2 camera. Where and how is it attached?
Logs show trying to connect to a mjpeg camera.
Looks like a lot of attempts at updating, too.
Please try to re-image, use 20200203.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

kalex1114, 2020-03-11:
I used Raspbian Buster Lite 2020-02-13 as you suggested earlier.  and manually configured motioneyes using the link you provided.  
Camera is attached via USB right into the pi.
camera i'm using is ELP USB with Camera 2.1mm Lens 1080p Hd 
https://www.amazon.com/gp/product/B00KA7WSSU/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

kalex1114, 2020-03-16:
So this is interesting.  I went back to my motionOS motioneyes SD card but didn't connect it to blue iris and so far no disconnects and video is still present.  Is it possible that pi is being overwhelmed by streaming it to blue iris?  any setting i can adjust to get this working?

starbasessd, 2020-03-17:
Curious.
I wonder if is a CPU over-heat issue on the Pi.
What are your settings under Video Streaming?
Do you have heatsinks on the chips, or a fan on the case, or is it in a
case with easily removed lid for testing?
Do you see any warnings for throttling for low voltage?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

kalex1114, 2020-03-17:
no heatsinks on the chips and no fan but middle of the case is open
[image]
I'm using POE for power.
video streaming settings:
[image]
where would i see warnings for low voltage?  I don't have a screen hooked up to it.  can you point me to which logs to check?

kalex1114, 2020-03-17:
Found it.  no under-voltage messages.  I did find them for one day only back on 9/27 thats when i had it plugged in via usb cable instead of POE and was doing something.  Other then that log is clean

starbasessd, 2020-03-17:
Here's a way to monitor CPU and GPU temps.
https://www.cyberciti.biz/faq/linux-find-out-raspberry-pi-gpu-and-arm-cpu-temperature-command/
Will try to get you a script in the AM (I'm EDT US East Coast time) to do
some logging...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

kalex1114, 2020-03-17:
Thank you. I’m on east coast too

starbasessd, 2020-03-17:
To log the temperature:
ssh into the Pi (user=admin, password='blank by default, or whatever you
set it to'

cd /data/usr <enter>
nano pi_temp.sh <enter>

-----  copy and paste  -----:
#!/bin/bash
# Script: pi_temp.sh
# Purpose: Display the ARM CPU and GPU  temperature of Raspberry Pi 2/3
# Author: Vivek Gite <www.cyberciti.biz> under GPL v2.x+
# -------------------------------------------------------
cpu=$(</sys/class/thermal/thermal_zone0/temp)
echo "$(date) @ $(hostname)"
echo "-------------------------------------------"
echo "GPU => $(/usr/bin/vcgencmd measure_temp)"
echo "CPU => $((cpu/1000))'C"
-----  end copy and paste  -----

<ctrl> x
y
<enter>

crontab -e

-----  copy and paste  -----
* * * * * /data/usr/pi_temp.sh >> /data/log/dmesg.log #every minute
-----  end copy and paste  -----
change first '*' to /2 for every other minute, /5 for every 5 minutes, etc.

<ctrl> x
y
<enter>

reboot <enter>

Now, it will  log the CPU and GPU temps to dmesg.log, and allow you to view
from "Expert Settings"
if you ssh in, you can view instant temperature with:
/data/usr/pi_temp.sh <enter>
or
tail -f /data/log/dmesg.log

DO NOT leave in this state.
To turn off logging:
crontab -e <enter>
put a # at the beginning of the line.

<ctrl> x
y
<enter>
reboot <enter>

Let me know what it shows.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

kalex1114, 2020-03-17:
done.  running now.  enabled blue iris to stream from motioneye and tracking temp.  so far same as before CPU/GPU are hovering around 60c

starbasessd, 2020-03-17:
Mine hovers around 45 no load, and 50 with 4 cameras (3 IP, one USB).
Might be the heat issue.

If it is thermal throttling, you can adjust (slightly) to 70C by adding a
line to /boot/config.txt
temp_soft_limit=70
There is a software throttle in the newer kernels that starts tripping at
60C.
It doesn't show as tripped until 80C.
This shows what the init settings are
vcgencmd get_config int
To see if it's hit 80C since last boot:
vcgencmd get_throttled

Let me know...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

kalex1114, 2020-03-17:
i didn't see any throttle messages in the logs
this is now:
[root@motioneye usr]# vcgencmd get_throttled
throttled=0x80000

camera still good.  will post results once it craps out

starbasessd, 2020-03-17:
Mine:
[root@meye-1acf1ec2 media]# vcgencmd get_throttled
throttled=0x0

I'll bet it's heat related. Watch the dmesg log for the temps. It's hit 80C
at some point since last reboot.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

kalex1114, 2020-03-18:
weird.  script is not logging.  i saw video feed stop and pulled temp.  it was 62.9C .  pulled log file but don't see temp readings in there
Log attached.
[dmesg (1).log](https://github.com/ccrisan/motioneyeos/files/4350740/dmesg.1.log)

starbasessd, 2020-03-18:
In crontab -e change
/data/log/dmesg.log
to
/data/log/temperature.log
Also you are still using the DEV version:
---- booting motionEyeOS dev20200223 ----
change to the 20200203 version, please.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-03-18:
And you are definitely getting toasty...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

kalex1114, 2020-03-18:
fixed the issue with crontab.  it should have been */2 not /2.  its loggging now

just checked for updates and no new versions are found:
motionEye is up to date (current version: dev20200223)

starbasessd, 2020-03-18:
Oops, probably a copy/paste error.
Go manually download and re-image
https://github.com/ccrisan/motioneyeos/releases/tag/20200203

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

kalex1114, 2020-03-18:
my version is newer i think should i downgrade?

starbasessd, 2020-03-18:
Yes, please.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

kalex1114, 2020-03-18:
any recommendations on cooling solution to get?

starbasessd, 2020-03-18:
Depends more on which case you have. I prefer the PiMoroni
shim, myself.
https://learn.pimoroni.com/tutorial/sandyj/getting-started-with-fan-shim
You don't have to run their software, but that's a nice bonus...
Sometimes you can get away with just heatsinks, if you have a fairly open
case.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

kalex1114, 2020-03-18:
I'm using POE hat so i might try this outhttps://us.amazon.com/gp/product/B082ZJYCS3/ref=ox_sc_act_title_1?smid=A2IAB2RW3LLT8D&psc=1

starbasessd, 2020-03-18:
> I'm using POE hat so i might try this outhttps://
> us.amazon.com/gp/product/B082ZJYCS3/ref=ox_sc_act_title_1?smid=A2IAB2RW3LLT8D&psc=1

For PoE I don't use hats, I use:
https://www.amazon.com/UCTRONICS-PoE-Splitter-Gigabit-Raspberry/dp/B07CNKX14C

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

kalex1114, 2020-03-19:
Thank you.  Ordered both to try out.

kalex1114, 2020-03-22:
Got pimoroni installed. wow what a difference.  Thank you
[root@motioneye usr]# ./pi_temp.sh 
Sun Mar 22 18:01:16 EDT 2020 @ motioneye
-------------------------------------------
GPU => temp=35.9'C
CPU => 35'C
[image]

starbasessd, 2020-03-22:
Awesome! Let me know if your overall issue is resolved, too.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

kalex1114, 2020-03-23:
Crap.  happened again.  Temp never went above 38c and currently at 32c

starbasessd, 2020-03-23:
Running for a long time at high heat can cause damage to the CPU/GPU &
networking chips, too. I know in the server rooms, we panic if the temps
get much higher than 40-50C which is why server rooms are kept so cold.
RPi Foundation made it high priority, when the Pi4 came out to issue
modified firmware to help cool it down, so quickly.
I'm sorry, but at this point, I'm at a real loss for more Ideas.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

kalex1114, 2020-03-23:
same.  might try new pi