motioneye-project/motioneyeos #2319: raspberry pi zero v1.3 board

jaynaidoopr, 2020-03-06:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

jaynaidoopr, 2020-03-06:
I have a raspberry board zero v.1.3, can it work on motioneyeos. I tried too many times burning an image and adding wpa supplicant conf to boot. and my configs are correct. When i check logs it says not connected to wifi and says cant find device.

starbasessd, 2020-03-06:
You didn't answer any of the questions above.
Can you image an SDCard with Raspbian, with teh wpa_Supplicant.conf to see
if that works correctly?
It looks as if the V1.3 is a clone of the V1.0, and does not have on board
WiFi: https://www.cytron.io/p-raspberry-pi-zero-v1.3
If that is the case, it will need a USB WiFi Dongle to work with WiFi.
It can still be used 'stand alone' as-is, but no Web GUI accessable unless
you use a USB to Ethernet or USB WiFi dongle.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?