motioneye-project/motioneyeos #2320: There is no email notification setting on my motioneyeos

danialaris, 2020-03-08:
[image]

How to fix it?

starbasessd, 2020-03-08:
Which version?
What networking settings? (WiFi or Ethernet?)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

danialaris, 2020-03-08:
I'm using 20200203, using WiFi

starbasessd, 2020-03-08:
I am unable to reproduce on either a Pi3b or ZeroW with a USB generic USB
WebCam & 20200203. Both give me full options when logged in as Admin on
fresh install.
If you are using a Pi4B, Pi3B+, Pi2B, or Pi1, I can test on them, also.
You might want to re-image as a first step.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

danialaris, 2020-03-08:
I'm using 20200203 on a Pi3B+ and a 5MP pi camera. I've already re-image it before and at first there was the motion detection setting and email notification setting, after I apply some setting, and certain setting ask for a reboot. The setting was gone.

starbasessd, 2020-03-08:
"Normal" Settings:
[image: image.png]
I just realized, you have Fast Network Camera turned on. When you do so, it
removes a lot of functionality, basically turning your device into a
Streaming Camera only.
Turn it off, they'll come back.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

danialaris, 2020-03-08:
Great. Done and Thank You!

closed 2020-03-08 by danialaris