motioneye-project/motioneyeos #2322: no dhclient after boot without wifi

dev246, 2020-03-10:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190911

### Board Model

I am using the following board/model: . Raspberry PI 3B+

### Camera

I am using the following type of camera: MMAL

### Network Connection

WiFi

###############################

When motioneyeos boot up without WiFi connection, and if WiFi connection come back, iwconfig show that we are connected to network, and now should be execute dhclient to obtain Ip address but this is not happening. Only after manually execute dhclient i got IP.

starbasessd, 2020-03-10:
Can you consider setting a static_ip.conf as a work-around until @ccrisan
or someone else can look at it?
Use whatever address scheme needed in sample. static_ip.conf goes in the
/boot partition, like wpa_supplicant.conf

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ccrisan, 2020-03-15:
@dev246 motionEyeOS uses a simplified networking mechanism that will only attempt to connect at startup and, if all connections fail, will reboot. If this is not suitable for your network, please use Raspbian with motionEye.

closed 2020-03-15 by ccrisan