motioneye-project/motioneyeos #2323: No umount after diconect USB device

dev246, 2020-03-10:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190911

### Board Model

I am using the following board/model: . Raspberry PI 3B+

### Camera

I am using the following type of camera: MMAL

### Network Connection

WiFi

###############################

After connect USB device automount works fine. But when i disconnect that device  old mount point remains. After connect this same device again a new mount point (/data/media/sdb - sdc -sdd - sdx) is created automatically with new device /dev/sdx. 
So after disconnect usb device i must manually umount it or reboot to be able connect it again and use configured storage path.

starbasessd, 2020-03-10:
It's always a good idea to either properly eject USB Drives, or power
off/remove. When you disconnect a USB drive while something is actively
pointing to it (File Storage), the system is going to be upset.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

dev246, 2020-03-11:
I known that we should property eject USB devices, I mean a situation where someone accidentally unplugged the usb drive, and would like to plug it in again. Now for everything to work it needs to be restarted, because the disk will be installed under a different path, the old one will not be available, so recording to disk will not work until reboot. Of course, currently used files will probably be damaged. 
The workaround is to add a script to the cron that executes every secoind with  mount command and if there is something mounted in /data/media/ sdx1 it checks if path /dev/sdx1 exist , if not then umount / dev / media / sdx1

starbasessd, 2020-03-11:
It would also have to check that the path to File Storage is, in the event
that someone plugs a different USB drive in, valid. Check that the USB
drive is valid (hasn't been corrupted by the unplugging), notify the system
that it's no longer a valid location in that event, etc. Another option
would be like 'Network Watch', and have the system force a re-boot, and do
all the testing.
As an alternative, to keep it viable, Move storage to the SDCard, notify
user/system that it's happened, to keep the ability to record viable.
So many issues... It might be easier to use a NAS location for storage, or
secure the USB drive to prevent 'accidental' removal. Even a Label-maker
piece of tape saying "DO NOT REMOVE"

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

dev246, 2020-03-11:
Network watch is uselessif you have camera for expamle in wood and you want only occasionaly download stored recording, no you donn't have network for most of the time.
I sow solution to that problem in usbmount package using udev rules.

starbasessd, 2020-03-11:
Have you thought about setting up as an Access Point? That way, it is
stand-alone, WiFi enabled to allow you to download the pictures/videos as
needed, without unplugging the USB stick. I attached the config files to do
it, if your PI has built in WiFi or a USB WiFi Dongle.
Several people on here have tried it for nature cameras, and it works well.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

dev246, 2020-03-11:
starbasessd  don't get me wrong, I noticed something that seems to be classified as a bug, not serious but always, so I reported it. According to the rule, not reported bug  = no bug . As you noticed, I also proposed a bit not elegant but simple workaround. If it gets corrected in proper way someday, that's great if not nothing happened , As you mentioned, you can always use red tape with the inscription Do not remove.

RTR75, 2020-05-16:
Hi starbasessd, as a newbe to motioneye and pi, can you point me to a nature camera setup, mostly setting up storage on a usb drive.
Thanks

starbasessd, 2020-05-16:
Biggest issue is power. Most any camera with IR (nightvision) works.
I've seen the 'official' Pi  Cam work well with a PiZeroW and set up as an
Access Point if you don't have nearby wireless.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

RTR75, 2020-05-16:
No need for IR camera, using a NoIr camera, to shut off just after dark and restart just before sunrise. The setup will be checked every day if not every other depending on the battery of course. 
My problem is getting the correct path set up to save the data to a USB. I have looked but all i can find is, do this and it works. My problem is I am not a great computer guy and I get lost very easy, especially between the "do..this..and..it..works. Do you know any good tutorials on setting up storage onto a USB drive so I can get from here to there?
Thanks for your input!

starbasessd, 2020-05-16:
Motioneye uses a read only filesystem.
If you want to make changes or store files, part of the file system free is
read writeable.
That part is under /data.
So, the default location for storage is /data/output/Camera<#>.
If you have a USB drive inserted during boot,  it normally shows as
/data/media/sda1 and you can store there instead.
Help any?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

RTR75, 2020-05-17:
yes that did, obviously i was trying to read to much into how to accomplish this small task. 
Thanks a LOT!

starbasessd, 2020-05-17:
You're welcome!
Feel free to contact me directly if you need more assistance.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

RTR75, 2020-05-17:
"Feel free to contact me directly if you need more assistance."
That would be great!
THANKS!

closed 2022-01-19 by starbasessd