motioneye-project/motioneyeos #2325: extremely poor performance

Insprill, 2020-03-14:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: Raspberry PI Zero W .

### Camera

I am using the following type of camera: MMAL Camera.

My camera model is: Not sure model number so here is a link to it: https://www.amazon.com/gp/product/B073HYJDCM/ref=ppx_yo_dt_b_asin_title_o00_s01?ie=UTF8&th=1.

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi.

Hello! So i've been at this for awhile now and can't figure it out. So the performance is extremely poor, like <=0.5fps and 30+ seconds of lag. Pretty much unusable. I tried everything, read every forum post that was similar, read through every GitHub issue on here that relates and can't find anything. And it stays the same whether the res is at 320x200 or if its at 1280x720. I disabled still pictures and it help a _small_ bit but not really noticeable. I also tried disabling recording and it made no difference. Now, i just enabled Fast Network Camera and bam. its working fine, no issues, except i need to record (continuous) and have them uploaded to Google Drive, which as far as i'm aware you can't do with Fast Network Camera on. I'm sorry if this is a stupid question but i really can't find anything, any help would be great :D

Insprill, 2020-03-16:
anything?

starbasessd, 2020-03-16:
I have never been able to reproduce 1080p 30fps with any PI CSI camera on a
PiZero. Ever. Any OS. Any Application.
It is a single core ARM processor, with 512MB RAM, a (now) low end GPU, on
a 700-900MHz clock.
Mine has done 1-5fps either recording video or pics, but not both. I also
use several IP cams, dedicated purpose, that get 10-15FPS to a NAS, for
either pics or movies, but not both. if using WiFi. (Better response using
Ethernet) Not much better (relatively speaking) on a Pi1, Pi2, Pi3B or 3B+.
The 3s get better frame rates (up to 10-15fps). I like them because they
are customizable as to where and how captured video and pics are stored. I
refuse to get a NEST or RING device because I A) don't want to pay the
monthly fees, and B) I don't want to give someone else my videos.
I have posted a Q: on their product page to see under what circumstances
they can claim their specs. They are misleading, at best.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Insprill, 2020-03-16:
No i can't even get 200x300 at 2fps

zubzero689, 2020-03-17:
Most of us if not all only use the zero w for fast network to another more powerful pi or other powerful single board computer.  Me I can barely get 6-7fps when fasting it to a pi4 or even less to an xu4 right now unless I use the .avi format that will do 10fps at the sack of some quality

jasaw, 2020-03-17:
@Insprill Pi Zero W can record (using h264_omx encoder) 720p 5 fps locally, and capable of running fast network cam mode RTSP at 1080p 30fps. I'm running 5 Pi Zero W cameras here and that's the performance I get. I even underclocked my Pi Zero Ws to reduce the operating temperature of the units. If you have a web browser connected to it, it will use CPU to encode MJPEG for live stream to the web browser. That's when the Pi Zero W struggles. Make sure you don't have a web browser connected to it then check what frame rate you're getting on the recorded videos.

Alternatively, you could set your Pi Zero W to fast network cam RTSP mode, stream to a more powerful device that does the motion detection and recording and uploading.

grimkor, 2020-06-24:
@jasaw Do you have the config you're using on both the Pi W and the host consuming that stream? I've been playing around with RTSP @1080p with mixed success and could use some direction. Currently my testing is by consuming the RTSP feed with VLC.

jasaw, 2020-06-26:
I have my RPI Zero W running at 1280 x 1024 @ 30 fps, bitrate 1 Mbps, image quality 25%, RTSP mode, MJPEG resolution 800x480 @ 5 fps, MJPEG bitrate 800kbps. They are all sitting within 30 meters from their nearest wifi access points.
I find that having multiple video streams going across wifi can degrade the frame rate of your RTSP streams. Try with one first, see where the limit is. Most likely your bottleneck is the wifi.

closed 2025-01-27 by Insprill