motioneye-project/motioneyeos #2327: MotionEyeOS USB camera not working anymore (was previously working)

subash774, 2020-03-15:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190911

### Board Model

I am using the following board/model:  Raspberry PI 3B+

### Camera

I am using the following type of camera: V4L2 USB camera 

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

### THE ISSUE
I was on the same version previously and everything was working perfectly fine. I had a problem with my broadband once and the stream showed still image (didn't update the live feed) so I rebooted the Pi. Ever since, the camera doesn't show up on V4L2 list.

When I try ```vcgencmd get_camera``` , I get ```supported=1 detected=0 ``` and ```/dev/video0``` doesn't exist either. It just suddenly stopped recognising the camera, nothing has changed. I even re-flashed the motioneyeos image onto my SD card and restarted the whole process, no change. 

Any help would be appreciated. 

Thanks!

starbasessd, 2020-03-15:
Have you attempted to use the camera in Raspbian, or connected to a PC or
MAC?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

subash774, 2020-03-15:
> Have you attempted to use the camera in Raspbian, or connected to a PC or MAC?
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

Yepp, tried and tested, the camera works fine on PC and Mac.

starbasessd, 2020-03-15:
Do you have the ability to SSH to the motionEyeOS Pi, or connect a keyboard
and mouse to the Pi?
Also, please attach the dmesg and boot logs (under Expert Settings)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

subash774, 2020-03-15:
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/4334998/dmesg.log)

Yeah, I can SSH into it and everything, in terms of OS, all is fine, the only problem is the USB camera that worked fine on the same release of the OS has now stopped recognising the camera.

starbasessd, 2020-03-15:
Can you re-image, and use 20200203?
https://github.com/ccrisan/motioneyeos/releases/tag/20200203

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

subash774, 2020-03-15:
I shall try it tomorrow and let you know, thanks a lot.

subash774, 2020-03-17:
@starbasessd Tried with the pre-release image, same result. 
[image]

dmesg log attached. 

Thanks a lot for your help :)
[dmesg (1).log](https://github.com/ccrisan/motioneyeos/files/4345123/dmesg.1.log)

starbasessd, 2020-03-17:
At this point, I'm going to have to give up. @ccrisan and @jawas may have
more ideas.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ccrisan, 2020-03-17:
That looks like a hardware problem. Check that the camera device actually works on that board.

subash774, 2020-03-17:
@ccrisan I also switched the SD Card with MotionEyeOS to another board (3B+ as well), same problem. It's baffling how it was just working absolutely fine and now it's suddenly stopped working and camera is working fine when tested on PC as well as OpenCV python

ccrisan, 2020-03-17:
If you change the board, the video device path might also change. I recommend either manually removing `/data/etc/camera*.conf` or flashing the OS image from scratch.

subash774, 2020-03-17:
After @starbasessd suggested, I flashed same and different versions of the OS, same result. I am thinking about using MotionEye with Raspbian, see if that works.

vincens2005, 2020-04-21:
I have the same problem

subash774, 2020-04-21:
Unfortunately, I haven't been able to solve it yet, do let me know if you find a solution

vincens2005, 2020-04-21:
Turns out the problem for me was that the pi's USB port failed and no keyboard or mouse worked with Raspbian on that board.

erew123, 2020-05-01:
Not sure if this is relevant. I have a Pi 3 A.... On a base image of MotionEyeOS... if I set the "Captured Before frames" much higher than 30 (cant give an exact number, though I know Ive tried 90), the camera turns off, its no longer detected/no image is visible, even on a reboot of MotionEyeOS.... 

I diagnosed by setting up a new SD card image and when I got to that setting, if I set it too high, the camera image is gone.... if I dropped the setting back down to something like 10, the camera image re-appeared/started working again.

starbasessd, 2020-05-01:
It's distinctly possible, that to buffer the last > whatever number of
frames, it runs out of resources (memory) and video fails.
If I get a chance this weekend, I can test. I can try a Pi3B+ and a
Pi4B-4GB and see if there is a significant difference.
What say you @ccrisan and @jasaw?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-05-02:
Test Setup #1
RPi3B, Tiga USB Generic USB Webcam, motionEyeOS 20200203-Pi3 8GB SD Card
Set Captured Before Frames incrementally from 10 - 100. in steps of 10
Waited for motion triggered recording & pics
It recorded, on average, about 3 minuted of video per motion, and a ton of
pics at the higher numbers
Was unable to reproduce.
Please send all your settings, and attach all you logs, and note at what
time you noticed the no image issue
Also, if you moved the number of Captured Befor Frames back below 10, was
the image restored?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

erew123, 2020-05-02:
Mines a Pi3a+... so I think 1/2 the memory of a Pi3B...which Im guessing is the issue in my instance. Personally, I have a happy medium for what I need, though I just thought Id mention it being an issue that seems to kill the camera.

When I put the settings too high, my system says:
[image]

And then on a refresh, the camera is blank/black/a placeholder
[image]

As for settings:

System: Pi3a+ (512mb of Ram)
Camera: Pi original camera 1.3
SSH and FTP are turned OFF
Camera is at 1900x1080, rotated and on 30fps
Text overlay, scale 4
Streaming set to 25Fps at 50% quality
Image resizing+Motion Optimization are OFF
Still images are on Manual Only
Movies are motion triggered at 75% quality with a max of 120 seconds.

Motion detection.... 0.2%
Auto threshold tuning OFF.
Auto noise detection ON
Light Switch Detection at 50%
Despeckle ON
Motion Gap 20
Captured BEFORE (What we are discussing, but fine at 10 and ? higher)
Captured After 30
Minimum Motion Frames 2
Mask ON and Editable
Show Frame changes ON

I believe that its motion re-starting after making the change and being unable to allocate enough memory (in my case) though thought may be relevant to other people... maybe asking their system to do too much?

This is in the dmesg (at the time of changing the settings... I think)
```
[ 8716.496845] Out of memory: Kill process 24553 (motion) score 749 or sacrifice child
[ 8716.496882] Killed process 24553 (motion) total-vm:296928kB, anon-rss:184404kB, file-rss:0kB, shmem-rss:0kB
[ 8716.527466] oom_reaper: reaped process 24553 (motion), now anon-rss:0kB, file-rss:0kB, shmem-rss:0kB
```

```
2020-05-02 17:04:52: [motioneye]    ERROR: motion failed to start
Traceback (most recent call last):
  File "usr/lib/python2.7/site-packages/tornado/web.py", line 1548, in _stack_context_handle_exception
  File "usr/lib/python2.7/site-packages/tornado/web.py", line 1756, in wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 290, in post
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 205, in wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 522, in set_config
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 373, in set_camera_config
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 507, in check_finished
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 482, in finish
  File "/usr/lib/python2.7/site-packages/motioneye/motionctl.py", line 129, in start
Exception: motion failed to start
2020-05-02 17:04:52: [motioneye]    ERROR: 500 POST /config/0/set/?_=AAAAAAAAAAAAAAA&_username=admin&_signature=AAAAAAAAAAAAA (192.168.1.248) 2365.79ms
2020-05-02 17:04:52: [motioneye]    ERROR: motion not running, starting it
2020-05-02 17:04:54: [motioneye]    ERROR: failed to start motion: motion failed to start
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/server.py", line 325, in checker
  File "/usr/lib/python2.7/site-packages/motioneye/motionctl.py", line 129, in start
Exception: motion failed to start
2020-05-02 17:04:54: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2020-05-02 17:04:54: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2020-05-02 17:04:54: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2020-05-02 17:04:55: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2020-05-02 17:04:55: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2020-05-02 17:04:55: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2020-05-02 17:04:55: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2020-05-02 17:04:55: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2020-05-02 17:04:56: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2020-05-02 17:04:56: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2020-05-02 17:04:56: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2020-05-02 17:04:56: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2020-05-02 17:04:56: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2020-05-02 17:04:57: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2020-05-02 17:04:57: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
```

starbasessd, 2022-01-19:
No comments after more than 60 days. If you are still having issues, please open a new issue.

closed 2022-01-19 by starbasessd

wrsmcode, 2023-01-14:
I can now confirm that version 20200203 [https://github.com/ccrisan/motioneyeos/releases/tag/20200203](url) works with my USB cam (Logitech C310)
The latest version did not work for me either. After this hint it works now, thanks for that

astinsan, 2024-07-24:
its power man.. use a powered switch or powered adapter.