motioneye-project/motioneyeos #2328: Continuous recording issue - recording stops after first clip

olmelabs, 2020-03-23:
### motionEyeOS Version
motionEye Version  0.42
Motion Version  4.2.2+gitUNKNOWN
OS Version  motionEyeOS 20200203

### Board Model
Raspberry PI 4 Model B 4GB

### Camera
Network Camera (RTSP). 
My camera model is: Dahua IPC-HFW2431T-ZS connected via POE switch

### Network Connection
My motionEyeOS unit is connected to the network via:  WiFi / Ethernet works the same.

### Peripherals
Not really, just saving videos on SD card for now. 

### Log Files
I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

[boot.log](https://github.com/ccrisan/motioneyeos/files/4369040/boot.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/4369041/motion.log)
 
Hi 
I am setting Movies to Continuous Recording with max length 300 seconds. The problem is that it stops recording after first movie when you press Apply or Reboot system.

Seems like i am missing something very easy.

Any help appreciated
Thanks a lot.

starbasessd, 2020-03-23:
Trying to reproduce.
Can you post motioneye.log, and messages.log, please?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-03-23:
@ccrisan/motioneyeos <motioneyeos@noreply.github.com>
I can confirm.
Using both PiZero & Pi4 4GB
20200203
I'm using WiFi connected Simple MJPEG camera
Movies: On
   Movie Passthrough: On or Off
   Recording Mode: Continuous Recording
   Maximum Movie Length: 60 seconds & 600 seconds
   Preserve Movies: For One Day
Motion Detection: On or Off
Records 1 segment.

Motion.log (entire):
[0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file
/data/etc/motion.conf
[0:motion] [NTC] [ALL] config_camera: Processing camera config file
camera-1.conf
[0:motion] [NTC] [ALL] motion_startup: Logging to syslog
[0:motion] [NTC] [ALL] motion_startup: Motion 4.2.2+gitUNKNOWN Started
[0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1:ml1:Camera1] [WRN] [NET] netcam_next: called with no data in buffer
[1:ml1:Camera1] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
MotionEye.log (last hour)
2020-03-23 08:10:38: [motioneye]    ERROR: failed to create movie preview
for /data/output/Camera1/2020-03-23/08-07-24.mp4: Command '[u'ffmpeg',
u'-i', u'/data/output/Camera1/2020-03-23/08-07-24.mp4', u'-f', u'mjpeg',
u'-vframes', u'1', u'-ss', u'4', u'-y',
u'/data/output/Camera1/2020-03-23/08-07-24.mp4.thumb']' returned non-zero
exit status 1
2020-03-23 08:30:51: [motioneye]  WARNING: mjpg client connection for
camera 1 on port 8081 is closed
2020-03-23 08:43:35: [sendmail]     INFO: sending email
2020-03-23 08:43:39: [sendmail]     INFO: email sent
2020-03-23 08:43:46: [motioneye]    ERROR: failed to create movie preview
for /data/output/Camera1/2020-03-23/08-43-18.mp4: Command '[u'ffmpeg',
u'-i', u'/data/output/Camera1/2020-03-23/08-43-18.mp4', u'-f', u'mjpeg',
u'-vframes', u'1', u'-ss', u'4', u'-y',
u'/data/output/Camera1/2020-03-23/08-43-18.mp4.thumb']' returned non-zero
exit status 1
2020-03-23 08:59:13: [sendmail]     INFO: sending email
2020-03-23 08:59:16: [sendmail]     INFO: email sent
2020-03-23 09:03:16: [motioneye]    ERROR: invalid literal for int() with
base 10: '79.2'
Traceback (most recent call last):
  File "usr/lib/python2.7/site-packages/tornado/web.py", line 1548, in
_stack_context_handle_exception
  File "usr/lib/python2.7/site-packages/tornado/web.py", line 1756, in
wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 1407,
in get
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 205,
in wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 1472,
in preview
  File "/usr/lib/python2.7/site-packages/motioneye/mediafiles.py", line
837, in get_media_preview
ValueError: invalid literal for int() with base 10: '79.2'
2020-03-23 09:03:16: [motioneye]    ERROR: 500 GET
/movie/1/preview/2020-03-23/08-58-56.mp4?height=79.2&_username=admin&_signature=9baa6a7876d12adebd5babef34e6394ac0c114bd
(192.168.2.224) 305.47ms
2020-03-23 09:09:59: [sendmail]     INFO: sending email
2020-03-23 09:10:02: [sendmail]     INFO: email sent

I am at a loss.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

olmelabs, 2020-03-23:
Thanks for looking into this.
Please find logs  attached
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/4371606/motioneye.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/4371607/messages.log)

ccrisan, 2020-03-23:
@starbasessd your traceback indicates a(nother) error that I have just fixed in https://github.com/ccrisan/motioneye/commit/3b402479622f780404b9dd7c97e6bf2c2c1158fb. It may happen that preview dimensions are supplied as float values.

ccrisan, 2020-03-23:
@olmelabs your `motioneye.log` indicates a rather large number of restarts, indicating a possible problem that is not related to motionEye configuration. If you're manually restarting it, leave it on for a couple of hours with Expert Settings > Enable Debugging on and let's see what the logs have to say then.

starbasessd, 2020-03-23:
> a(nother) error that I have just fixed in ccrisan/motioneye@3b40247

Will this be in a new dev build, tonight?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ccrisan, 2020-03-23:
Not yet. It's a motionEye commit and it has to make its way into motionEyeOS first.

starbasessd, 2020-03-23:
Thumbs up,

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

olmelabs, 2020-03-24:
@ccrisan 
Yeah, i was restarting it many times trying different options etc.. So the level of restarts is quite high.
I did what you said this morning. Attaching files.
2020-03-24 entries are with Debugging On
Thanks
[motion.log](https://github.com/ccrisan/motioneyeos/files/4374814/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/4374815/motioneye.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/4374817/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/4374818/dmesg.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/4374819/messages.log)

ccrisan, 2020-04-12:
@olmelabs can you try the exact same setup using a different camera (type)?

There's nothing that I could find in your logs that would indicate recording to stop after the first clip. It might be though a configuration issue or a bug in Motion. The former would result in the same problem with any other camera type, while the latter should probably go away with a different camera.

starbasessd, 2020-04-12:
I have USB and Network mjpeg cameras, and could do fresh installs on
PIZeroW, Pi3B+, and Pi4 for testing with clean logs.
Do you have the other commit in motionEye or motionEyeOS for testing yet?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

olmelabs, 2020-04-29:
Plugged in my USB cam on fresh image.
No camera settings change - only switched on continuous recording.

Only added WIFI and stopped FTP and SMB servers.
Same - only first movie recorded.
After reboot - also only one movie.

Just in case - logs
[motion.log](https://github.com/ccrisan/motioneyeos/files/4552218/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/4552219/motioneye.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/4552221/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/4552222/dmesg.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/4552224/messages.log)

fudgeboy2000, 2020-05-16:
Same issue here - Pi3B+, network cam, continuous recording generates one file and then nothing else.

polvadis, 2020-05-21:
I have the same problem (Pi4 4gb) and found this thread while looking for an answer. Seems like the bug is ongoing though. Using a USB camera over WiFi.

dannysuls, 2020-06-11:
I upgraded from 20190911 to 20200606 yesterday, and now I also have this problem. The logs give me no clue what's causing this.
This is a showstopper for me, I guess I will have to downgrade to 20190911 again. :-( 
Hopefully this will get solved with the next stable release...

kaeser-sosae, 2020-06-17:
Same issue here, Rpi 4B, only records one clip, regardless of how long you set it for. Have tried many settings to fix this but cannot get it to record more than one clip.

Attreyoo, 2020-06-17:
Yep, Same here.  Every version of 2020 seems to have this issue. Please help.

kaeser-sosae, 2020-06-17:
I can confirm that multiple cameras experience the same issue. I have 1x Rpi camera connected and 1x network camera connected over rtsp - same issue on both, only records the first clip.

I've tried very short clips - 60 seconds - and very long clips such as 24 hours and it stops at one clip regardless.

starbasessd, 2020-06-17:
Known issue. Current work-around is 20190911

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

kaeser-sosae, 2020-06-17:
Is it possible to downgrade or is it a reinstall situation?

starbasessd, 2020-06-17:
Re-image, unfortunately.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

TommyG007, 2020-07-12:
Same issue for me on 20200606. I'll temporarily re-image to 20190911. Hopefully the issue can be fixed soon.

camilomarino, 2020-07-12:
Same problem on raspberry pi zero w.

luissalcedo, 2020-07-29:
Same problem here!

starbasessd, 2020-07-29:
Known issue with all the 2020 images. Try the 20190911 images.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

pncrtx, 2020-08-05:
I have the same problem! I had to downgrade to 20190911

bernardflach, 2020-09-12:
Ive got the same issue with the 2020 images. Recording stiops after 1 clip. 
I also reverted to the 20190911 image. On this image the recording works

starbasessd, 2020-09-12:
Have you tried 20200907?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

bernardflach, 2020-09-12:
Not yet.... i shall try however im more into stable releases

bernardflach, 2020-09-12:
It was confirmed in this post by some users ... hope its fixed in a new stable release. otherwise i'll stay on 20190911

starbasessd, 2020-09-12:
No, 20200907 has been out less than a week, and most posts are older than
early August. I can test with you if you let me know your setup.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

bernardflach, 2020-09-12:
running 20200907. 
Rsan imediately into some issues : 
-added camera (once configured)  cannot be configured again.
- Changing some options does not trigger to show the apply button.

starbasessd, 2020-09-12:
Which RPi?
What camera? I just added a camera, deleted it, re-added. (actually 2, a
mmal and a network)
Need specifics of options, please.
I can test on PiZero, 1, 2, 3B, 3b+, 4-4GB & 4-8GB. I have several USB
cameras, several network cameras, and a PiCam v1.3 for testing.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

bernardflach, 2020-09-12:
Rpi 4, - 1 TB external HD, a foscam 2CM 
All of this is working on the 2019 image : lets say the recording works on this image perfetly.

On the lates release the recording does not work. (just 1 movie) nothing more

bernardflach, 2020-09-12:
hmmm... the bugs i mentioned earlier are brower related. the not appearing apply button works in firefox, not in Edge

starbasessd, 2020-09-12:
How are you ending the first recording?
What resolution?
Is your Foscam connected via RTSP?
Just trying to duplicate your setup.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

bernardflach, 2020-09-12:
Hang on , on firefox i can now set the proper settings ... its now recording ...  

I'll mail you in a while with all findings and details

Fact is , and ill post a bug for it, the tool does not work well anymore on Microsoft egde.

starbasessd, 2020-09-12:
FF is my default browser for motionEye/motionEyeOS due to the multiple
times Chrome broke after an update. Since the new Edge is Chrome-based, I'm
not surprised.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

fudgeboy2000, 2020-09-26:
I just upgraded to the 20200907 nightly and the continuous recording issue has been fixed for me!  Good stuff, many thanks.

weemscreekstudios, 2020-10-04:
> I just upgraded to the 20200907 nightly and the continuous recording issue has been fixed for me! Good stuff, many thanks.

What is the best way to upgrade to the nightly build? I'm using a Pi Zero. Do I need to re-image an SDCard?

weemscreekstudios, 2020-10-04:
> > I just upgraded to the 20200907 nightly and the continuous recording issue has been fixed for me! Good stuff, many thanks.
> 
> What is the best way to upgrade to the nightly build? I'm using a Pi Zero. Do I need to re-image an SDCard?

Never mind, I just discovered the Enable Prereleases switch under Expert Settings.

ramybt, 2020-10-05:
i have just upgraded from the 2019/9/11 to  version 20200907 after when i heard its fixed, it works and its recording but its slow and my both cameras have issue with the video it self, its showing green lines all over 
[image]
[image]

does anyone know why? can i fix it?

RPi3+  with 32GB for cam 1, 16gb cam 2
motion os 20200907
wifi

thank you

starbasessd, 2020-10-05:
I use (by default) H.264, or Movie Pass Through, and I do not have these
issues. In another thread, there is discussion about whether something
changed upstream in ffmppeg (used to transcode video).
Let me know if either help.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ramybt, 2020-10-07:
I tried no movie pass and the problem was solved but than my video start using a lot of memory space of my SDcard almost **3GB** per **30** minutes of recording
I been switching between different video formats and problem is same, its either **green lines** or fixed but using lots of **GB**.

I might just go back to the older version no problem with that one but if someone can help me here maybe, Thank You

lbismarck, 2020-10-30:
Any news on this? Still getting this problem on the latest official version. thx!

starbasessd, 2020-10-30:
20200606? Yes.
dev20201026 seems to be working for me on PiZeroW and Pi4-4GB

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

lbismarck, 2020-10-31:
@starbasessd thx, that seemed to work re recording! however now I only see a grayed-out screen when I open the preferences of my camera. cannot edit anything anymore eventhough I am logged in as admin....

starbasessd, 2020-10-31:
If you reboot the Pi, and you still have issues, try re-imaging the SDCard,
sounds like something got corrupted.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ITCMD, 2020-12-02:
I'm still having this issue on 20200606. First clip records then there are no more for each day.

starbasessd, 2020-12-02:
Please try dev20201026

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ITCMD, 2020-12-03:
Will do. Unable to check this for a while as I'm busy with work but I'll try next day off!

belese, 2020-12-14:
Hello, 
i've also this issue on the continous recording, only the first is recorded

motionEye Version | 0.42.1
Motion Version | 4.2.2+gitUNKNOWN
OS Version | motionEyeOS 20200606

i'll try the dev version as soon as i can.

starbasessd, 2020-12-14:
dev 20201026 or 20200911 (if not using a Pi4b or the new HQ PiCam)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

wa1id, 2020-12-21:
I was on 20200606 with the same issue. After updating to dev20201026 the issue is resolved.
I'm using a RPi 4 (4gb) with an action camera attached via usb 2.0 and a 1TB HDD usb 3.0

ramybt, 2020-12-21:
Late update, i have been using 20201026 and it's working great so far.

20201026 is working 100%

belese, 2021-01-05:
Look like it's working too for me with the dev Release.

ITCMD, 2021-01-06:
Awesome, glad to see it was looked at.  Once it is in an official release I will close this issue.

ITCMD, 2021-01-13:
Is there a way to update to a dev release (instead of a fresh install)?

starbasessd, 2021-01-13:
Expert Settings, Enable dev releases

closed 2022-01-19 by starbasessd