motioneye-project/motioneyeos #2329: Motion Detection starts during IR filter transition

RichBuck89, 2020-03-24:
**Preliminary Docs**
I confirm that I have read the CONTRIBUTING guide before opening this issue.
yes
I confirm that I have read the FAQ before opening this issue.
yes

**motionEyeOS Version**
motionEyeOS 20190911
motionEyeOS dev20191031

**Board Model (both having same issue)**
1 x Raspberry PI 4B
1 x Rasberry PI zero w

**Camera**
MakerFocus Raspberry Pi4 Camera IR, Raspberry Pi4 Camera Night Vision Fisheye Camera 5mp OV5647 IR-CUT
[LINK](https://www.amazon.com/gp/product/B07DPNDP6R/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1)

**Network Connection**
1 x gigabit Ethernet
1 x Wireless N

Hello, 

I have several cameras with the IR-CUT [filter](https://www.amazon.com/gp/product/B07DPNDP6R/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1)

They work great and remove the red tint during day time video. One issue I have is that the motion detection starts when switching between the IR filter randomly. 

I've found that changing the Minimum Motion Frames helps but setting it to 90 frames is around 5-6 seconds of missing motion detection any higher it seems to work but I have to stand there and do jumping jacks for a while to get the motion detection to start... I've played around with the Light Switch Detection and it doesn't seem to do anything even at 100%. 

Is there a better way of getting the IR-CUT to stop tripping the motion detection? 

Sample [video](https://drive.google.com/open?id=1cOO3jpyMCTXcfrV_5r5E6QKey-amxeQp)

Thanks, 
jake

starbasessd, 2020-03-24:
Have you tried Light Switch Detection?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

RichBuck89, 2020-03-25:
Yes, I've set it to 100% and tried a few others as well. Doesn't seem to help.

RichBuck89, 2021-10-06:
> I know the topic is old, but I have the same problem. Did you find any
> solution? Or as a workaround, were you able to disable the ir filter from
> the camera?

No I never found a solution.