motioneye-project/motioneyeos #2330: Question about correct configuration for best performance

kkwwm, 2020-03-25:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190911

### Board Model

Raspberry PI 3B+

### Camera

4 cheap Chinese IP wifi cameras. Each of them set as "Network camera"

### Network Connection

WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

SATA External Hard Disk

### Log Files

Hello Guys!
I'd like to ask about correct configuration for best performance. I'm using RB Pi 3B+. Motion detection is turn off. Continuous recording turned on. Resolution for each cam is 1280x720, frame rate set as 15fps. I have terrible lags on preview as well as on recorded movies. Real fps is maybe 0,5fps or less. Sometimes I have to wait 30 or more seconds to refresh the preview.
How should I configure this set to achieve the best performance?? Is the exchange RB Pi3B+ for Pi4 or other board will improve performance??
Thank you,
cheers!
Konrad

starbasessd, 2020-03-25:
What networking between camera and Pi?
What camera are you using?
Have you tried lowering resolution, either at camera or in motionEye?
(800x600, or 640x480?)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

kkwwm, 2020-03-26:
Cameras are connected via WiFi to the same (wireless) network as RB with motioneye. Those are some cheap Chinese ip cameras like this: https://www.aliexpress.com/item/4000085391518.html?spm=a2g0s.9042311.0.0.28134c4dS2YCYU
When I open the rtsp stream in the browser the stream is smooth and no delay. I can't lower the resolution because picture quality will be unacceptable.

jasaw, 2020-04-15:
I would try these:
- connect RPi to Ethernet rather than wifi.
- reduce the video frame rate and/or resolution and/or bit rate.
- upgrade to a more powerful device that is capable of recording 4 video streams concurrently.
- try with 2 concurrent video streams first, then increase to 3, then 4 to see where the limit is.