motioneye-project/motioneyeos #2332: Cannot Add Camera.. error in Camera field

LindsayFowler, 2020-03-27:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motioneyeos-raspberrypi3-20200203

### Board Model

I am using the following board/model: Raspberry PI 3B+.

### Camera

I am using the following type of camera: Remote MotionEye Camera.

My camera model is: knock off raspberry pi camera from eBay on the remote pi0w (which was able to be added in other versions)

### Network Connection

Both motionEyeOS units are connected to the network via: WiFi

### Peripherals
none relevant

### Log Files

none
[image]

 I used this dialog box in a previous version successfully, but in this latest pre-release version, the remote MotionEye Camera field will not be populated and so I cannot add the camera on my second pi (which is a raspberry pi 0W).

starbasessd, 2020-03-27:
Please answer all the questions to the best of your ability:
On the PiZeroW,
Which version or motionEyeOS are you running?
Can you see the stream on port 8081 from another computer?
Do you have Fast Network Camera on?
On the Pi3B+:
Have you tried adding :8081 to the 192,168,1,109: (also try adding http://
to make it http://192.168.1.109:8081)
Please let me know.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

LindsayFowler, 2020-03-27:
On the pi 0W it's motioneyeos-raspberrypi-20190911
On the 3A+  now it's motioneyeos-raspberrypi3-20190911 and still same problem.
(I know I said 3B+ originally - my mistake)

Tried adding the port number 8081, but not sure why I would need to do this as I have not changed the ports anywhere.
[image]
This did not help.
Also tried both Safari and Firefox web browsers.

LindsayFowler, 2020-03-27:
Ahh sorry - I see the protocol http:// is required!
[image]

To me it just looked like a broken text field as there was no feedback until the protocol was entered. I can see you did have the protocol in the example.

LindsayFowler, 2020-03-27:
Thanks so much for the swift and helpful response.

starbasessd, 2020-03-27:
You are welcome. Don't forget to close the ticket if you have it working.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?