motioneye-project/motioneyeos #2333: MotionEye takes a 800kbps H264 MPEG4 feed and turns it into over 22Mbps MJPEG Stream

rog889, 2020-03-28:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 

motionEye Version | 0.42
Motion Version | 4.2.2
OS Version | Raspbian 10

### Board Model

I am using the following board/model: 

Raspberry Pi 4

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: Hikvision Turret and Samsung Smartcam

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

Ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

rog889, 2020-03-28:
I am taking a simple RTSP h264 feed 15fps 1280x720, feeding motioneye at 800kbps, but if I use http MJPEG streaming at 15fps with 800x600, it turns out to be over 22Mbps video stream on the browser.  Why would motioneye convert something into so large?  its 22x!  Is it possible to make it a lot leaner like less than 1Mbps?

rog889, 2020-03-29:
Anyone can shed some light please

starbasessd, 2020-03-29:
@jawas?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

rog889, 2020-03-29:
I would love to attach screen shot to show both codecs info side by side. But it looks like this place doesn’t have this feature.

jasaw, 2020-04-15:
Simple explanation is H264 over RTSP is much more efficient at transferring video over the network because only changes between frames are sent, whereas MJPEG method sends every single frame as a full JPEG image. Your MPJEG @15fps @800x600 is actually 22Mbps of JPEG images. When you (pass-through) record at the receiving end, motion software just saves the video stream into file without any modification. If you want to receive MJPEG stream, it would be better to encode it into H264 format.