motioneye-project/motioneyeos #2334: Connecting to a VPN

AlephDev, 2020-03-31:
I have a VPS running an OpenVPN server. I want to connect my Raspbery Pi running motioneyeos to this VPN so that I can view the camera streams from beyond my home network.

Normally, on my linux machine, I open the networking app and click import VPN connection which is when I would open the client.ovpn file. 
 
I tried Googling how to pre configure/setup a vpn on a headless Raspberry Pi but I haven't had much luck by ssh'ing into the RPi running motioneyeos.

Any advice would be much appreciated.
Cheers.

starbasessd, 2020-03-31:
Not an option built in to motionEyeOS, and only option to get it in would
be to compile it into your own build.
Best bet would be to get Raspbian on the Pi, set up for the VPN, and then
install motionEye (not motionEyeOS).

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

AlephDev, 2020-04-02:
Thanks @starbasessd I will try that.

Do you by any chance know how I can access files on my home pc from my cellphone on 4G, using my VPS/VPN?

starbasessd, 2020-04-02:
That is not a discussion for this forum.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

AlephDev, 2020-04-02:
Might you know which forum it would be a discussion for?

starbasessd, 2020-04-02:
I would try an OpenVPN or VPS forum, personally.
Searched on google for access pc files from phone
First pop up was:
https://www.howtogeek.com/174301/how-to-access-a-windows-desktop-from-your-tablet-or-phone/

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2022-01-19 by starbasessd