motioneye-project/motioneyeos #2337: Fast Network Camera will not work w/ stream authentication on

Sasquiche, 2020-04-01:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: **20190911**

### Board Model

I am using the following board/model: **Raspberry Zero**

### Camera

I am using the following type of camera: **V4L2 Camera**

My camera model is: **Rasberrypi V2**

### Network Connection

My motionEyeOS unit is connected to the network via: **Wifi**

### Peripherals

**None**

### Log Files

[motioneye.log](https://github.com/ccrisan/motioneyeos/files/4414622/motioneye.log)
[streameye.log](https://github.com/ccrisan/motioneyeos/files/4414625/streameye.log)

### Issue

I've altered the default passwords for both admin and user. If I apply "basic" authentication to video streaming, while FNC is on and set to MJPEG, the motioneye web interface video (using firefox) is turned off. Streaming from "http://username:password@x.x.x.x:8081" brings up the authentication but does not accept any of the correct logins. What's even stranger is if I try to run the stream from VLC (x.x.x.x:8081), it will crash as soon as I run it.

I can switch authentication off and everything will work as usual. 

Not related to motioneye: I can confirm that VLC could handle the authentication requests perfectly fine when FNC is turned off, and video streaming authentication is turned on with "basic"

Sasquiche, 2020-04-01:
Just a note to anyone having VLC issues, according to [motion](https://motion-project.github.io/motion_config.html#OptDetail_Stream) VLC will not open a MJPEG stream. Which covers the VLC crash.

cesarandreslopez, 2020-09-26:
Can confirm same issue as above