motioneye-project/motioneyeos #2338: Do not store images on SD card, save only FTP upload

ralfberger, 2020-04-02:
### motionEyeOS Version
motionEye Version | 0.41
Motion Version | 4.2.2
OS Version | motionEyeOS 20190911
### Board Model
 Raspberry Zero W
### Camera
vc.ril.Camera (MMAL) RaspiCam

Hi,
I save the pictures via FTP to my server. Therefore I do not want to save the images on the SD card. Because of all the saving and deleting, the SD card does not last very long. How can I achieve that the images are only stored on my server and not on the SD card. 
Thanks and best regards

starbasessd, 2020-04-02:
Not at home right now, but can you try the path

/dev/null

for the local storage path?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ralfberger, 2020-04-02:
Thanks, that seems to work.
I tried a fake path before, but then I didn't have a live picture anymore.

starbasessd, 2020-04-02:
It's not a 'fake path'. It is just a black hole to dump things into. I
wasn't sure if there were any file checks that would mess it up while/after
dumping the data there.
Happy saving the SDCard.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ralfberger, 2020-04-02:
/dev/zero
was not a good idea,
Then no mail with attachment will be sent and no pictures will be uploaded via FTP. I have now entered this path again  
home/ftp/sdcard/Camera1
and mail with attachment and also FTP upload work

starbasessd, 2020-04-02:
So, if working the way it should, awesome.
Have a great day!

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ralfberger, 2020-04-02:
yes, it works, but now I have what I didn't want. The pictures are written on the SD card and that is exactly what I did not want.

starbasessd, 2020-04-02:
Where are you finding the files? Where are they getting written to?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ralfberger, 2020-04-02:
I did not want the images to be saved on the SD card when motion detection was activated, but only uploaded to my server via FTP and I received an email with an attachment
Unfortunately, it does not work with "/dev/null/".  Then I don't get a mail with attachment and it is not uploaded via FTP.
[[image]](url)

starbasessd, 2020-04-02:
My setup for testing (working:
PiZeroW / 8GB card
D-Link DCS-5020-L camera (Network camera selected)
File Storage
     Storage Device: Custom Path
     Custom Path: /dev/null
     Upload Media Files: On
    (Settings for my NAS FTP storage)
Motion Detect On
Still Images and Movies set to Motion Triggered

Yep, /dev/null doesn't work. It puts the file there, and then tries to copy
the file it just wrote over to the FTP server. Let me play.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ralfberger, 2020-04-02:
Don't worry, there's no hurry.  It would be nice if there was a solution.
Stay healthy

starbasessd, 2020-04-02:
If your FTP server supports Windows shares (SMB) you can change from Custom
Path to Network Share
This DOES write directly to the share and not to the SDCard.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ralfberger, 2020-04-02:
Unfortunately the server does not support SMB.

starbasessd, 2020-04-02:
This probably needs to be a request for feature, then.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?