motioneye-project/motioneyeos #2339: Maximum Movie length only applies to motion capture movies

awolmage, 2020-04-02:
### Issue

Maximum Movie length only applies to motion capture movies and not continuous recordings.

So I've set up a camera to monitor some activity around my house recently with the aim of sending the videos to the local police for a crime reference number I have open.

I can set Maximum Movie Length option to 0 and it will happily record indefinitely and create a new video file at random after it is several GB in size. If I set it to 600s (5 mins), for example, I would expect this to do continuous recording and every 600s create a new video file (this would make it far easier to download and send to another source as evidence).

The tooltip wording suggests at first that Maximum Movie Length applies to all movies but later suggests it only applies to Motion movies.

Please could you let me know if this is a bug, intentional/limitation, or if there's something I need to set in particular to make Continuous recording start a new video after a certain amount of seconds?

Thanks.

----------------------------------------------------------------------------------

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

motionEye Version | 0.42
Motion Version | 4.2.2+gitUNKNOWN
OS Version | motionEyeOS 20200203

### Board Model

Raspberry PI 3B+

### Camera

Camera Device |  vc.ril.camera
Camera Type |  MMAL Camera

Pi Cam v2

### Network Connection

WiFi

### Peripherals

N/A

### Log Files

N/A

JohanD78, 2020-04-07:
Already reported several times. Motioneye have pushed a fix for it but we will have to wait untill it finds its way into motioneyeos dev builds. 

See here for some info: https://github.com/ccrisan/motioneyeos/issues/2235

awolmage, 2020-04-07:
@JohanD78 Thanks for the additional info. Glad it's a bug and it's getting fixed. Was going a bit mad trying to figure out why this didn't work the way I thought most recording systems are set up.

Happy if this gets marked as a duplicate.

retardexpress, 2023-01-30:
i just ran across this same issue with continuous recording stopping completely after the preset # of seconds.. 
which is fine i can work with that
BUT
what im trying to figure out now, is on continuous recording, how large will the files become before a new file is started?
i'm assuming it wont just keep going on a single file till the SD card/drive is full?
my concern is, if its recording and power is suddenly interrupted, the ENTIRE video file may become corrupt

thanks for any thoughts about the file sizes!