motioneye-project/motioneyeos #2340: Feature request - Wi-Fi strength readout

GaryMorrisJr, 2020-04-02:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190119.

### Board Model

I am using the following board/model: Raspberry PI Zero W (and others).

### Camera

I am using the following type of camera: Simple MJPEG Camera.

My camera model is: Cheap Ebay V2.

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

None

No issues really but I do have a feature request, if it is possible.

It would be really helpful to know the signal strength of the Wi-Fi for each device. It could explain some issues that we experience.

I did search to see if this was suggested and did not see this.

Thank you for an excellent software package!

Gary

starbasessd, 2020-04-02:
Because motionEye uses busybox for many of it's functions, you don't get
everything a normal iwconfig has but

iwconfig wlan0 | grep -i quality

gives you signal strength and quality of signal from command line.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-04-02:
You could also use

watch -n 1 cat /proc/net/wireless

which shows status, strength, quality, dropped packets, etc continuously.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?