motioneye-project/motioneyeos #2341: HTTPS nginx reverse proxy

lukasig, 2020-04-03:
Hi

I have motioneyeOS running on a rpi3 and i can access it externally through DDNS.

   I am trying to get the connection secured, and as far as I understand it is possible but i can't figure out how.
   I have read [this](https://github.com/ccrisan/motioneye/wiki/Running-Behind-Nginx) thread. I can SSH my pi3, but commands such as apt-get are not available for MotionEyeOS.

I don't want to have to install MotionEye through raspbian. I want to use MotioneyeOS.

Can anyone point me in the right direction?
how can letsencrypt and certbot help?

thanks

starbasessd, 2020-04-03:
Best bet would be to BYOM: (Build Your Own MotionEye)
https://github.com/ccrisan/motioneyeos/wiki/Building-From-Source
Be aware, a lot of functionality is done with busybox instead of actual
individual apps.
motionEye also uses overlays, too, to protect the system as read only.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

lukasig, 2020-04-03:
I am not very advanced when it comes to the terminal. I definitely don't feel comfortable building my own... anything from source :)

I thought I could get a secure connection to my Motioneyos through something simple like copying and pasting a few commands and editing a few conf files...

starbasessd, 2020-04-03:
Your router may support VPN which could give you your secured connection to
your network, without putting the motionEyeOS on the internet...
You could pick up another Pi to make your own VPN server.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

digiltd, 2020-04-03:
Just to clarify, you want to be able to access your motioneye from any browser on the internet and you want to do so with a secure connection i.e. HTTPS?

LetsEncrypt is a service that provides the free SSL certificate, Certbot is the software used to create and manage the LetsEncrypt certificates. But you need to be able to install them and Motioneyeos doesn't let you. Best bet is to try the vpn suggestion, but be careful whatever you do as if you don't know what you are doing you could end up making all sorts of bits on your local network accessible to the internet without knowing it.

lukasig, 2020-04-03:
@digiltd you uderstood me well :)
Since the issue is not as simple as i predicted, let me put this another way:

I have a single raspberry pi 3 available. What is the most simple setup to be able to run motioneye on it and have secure access to it from any browser on the internet?

I think I might be able to handle installing motioneye through raspbian lite, but from there on i'm not exactly sure how to set up https.

is there any simpler way?

thanks again

starbasessd, 2020-04-03:
motionEye by itself isn't https:// capable. Any solution would need to
install either a VPN on Raspbian on your Pi3 (doable, but not elegant or
easy, with motionEye also on the same Pi3, or a https:// app that forwards
requests to the web page for motionEye.
As you can see, it's not going to be easy, no matter which path you follow,
but it IS doable, just not 'copying a few commands or editing a few config
files'...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

lukasig, 2020-04-03:
Based on the [comment](https://github.com/ccrisan/motioneyeos/issues/49#issuecomment-161096384) of @ccrisan am i right in assuming that my credentials are not leaked even though the connection is not secure or am I missing something?

starbasessd, 2020-04-03:
Installing wireshark, testing in a few.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-04-03:
I hate this time of year. New Ubuntu to be released this month, but the
last version has SOOOO many updates to apply on a new install..

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

lukasig, 2020-04-03:
I am testing my abilities to go a different route, but then i ran stumbled [here](https://github.com/ccrisan/motioneye/issues/1686).

lukasig, 2020-04-04:
so

I have managed to install motioneye in raspbian buster lite and proceeded to work on the secure connection. I have used the [instructions from certbot page,](https://certbot.eff.org/lets-encrypt/debianbuster-nginx):
`sudo apt-get install certbot python-certbot-nginx
`
`sudo certbot --nginx
`
and now when i access it externally i get a 'welcome to nginx page'

I am not sure what i need to do next...

lukasig, 2020-04-04:
As a sidenote

I have recently installed nextcloudpi on a rpi4 and setting up a secure connection was as simple as [pushing a button](https://lh4.googleusercontent.com/ic6X3iIY3U2IZcG8_qXvANkY8scqkk81gAMixauNi9Yj9ZMLcyho2Bw_0MfVn3940qeOh3DuCepJ4g=w1879-h961):

Is it possible to implement something similar in motioneyeOS?

lukasig, 2020-04-04:
I decided that for the moment the simplest way to get it over with is to use motioneyeOS accessed on a port of choosing other than port 80, and not log as admin externally.

But an implementation of a secure connection like the one in nextcloudpi mention above would be of great help for many of us I believe.

thanks again for your support

closed 2020-04-04 by lukasig

adorablyadam, 2023-10-14:
@lukasig how's it going on in this regard?
I have no need to view via external internet - but need a way to host and display locally on a tablet I have mounted. 

The latest android updates now ignore the allow mixed-content setting and so since the local stream is http and not https, it's now all being blocked. Was working fine for years up until now.

lukasig, 2023-10-15:
@adorablyadam 
I ended up installing the [dev](https://github.com/motioneye-project/motioneye/tree/dev)  branch of motioneye on a raspberry pi running dietPI and setting up a secure connection using caddy2, which was simpler than I expected.
Since then I have also upgraded to a x86 mini PC, and I run frigate on docker using ubuntu server. Google coral USB and frigate is a setup that i really like. And caddy still does the job very well.