motioneye-project/motioneyeos #2342: stream motioneyeos to facebook

Birdcam, 2020-04-03:
Hello,

I have been using Motioneyeos on an Raspberry Pi 3B+ with USB webcam to watch birds nesting. I would like to stream this live to facebook and/or YouTube. Is this possible? If so, how do I do this?
I am sorry to say I am not verry good at using Linux.
Thanks,

Louis

starbasessd, 2020-04-03:
Possible?  Yes.
With motionEyeOS, not any way I found.
You would need to run Raspbian with motionEye, then feed the stream, or
another PC to feed the stream
https://www.facebook.com/motioneyeos/posts/crisan-ive-managed-to-figure-out-how-to-stream-to-youtube-i-needed-to-run-motion/910079139101862/
discusses it in brief.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Birdcam, 2020-04-07:
After a lot of failure I found out that Facebook no longer supports rtmp streaming. Starting november 2019 they only support rtmps. Ffmpeg does not support that. Now I am only able to stream to YouTube and not to Facebook.
Any solutions anyone?