motioneye-project/motioneyeos #2343: No Network connection with usb ethernet adapter

thecheeseontoast, 2020-04-03:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190911

### Board Model

I am using the following board/model: Pi Zero (not W)

### Camera

I am using the following type of camera: can't even boot, let alone worry about the camera

My camera model is: "

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet, tp-link adapter

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * USB-ethernet adapter

##################

Pi is stuck in bootloop as the OS fails to find a network. Testing on raspbian, the network is found instantly with no set-up. Only on motioneyeos does it fail to find anything. As such, I can't get past the boot. Adapter works fine with raspbian, my linux machine etc. Status light indicates it has connection as motioneyeos boots, yet the os clearly can't find anything.

Any suggestions on steps I should take?

starbasessd, 2020-04-03:
Do you have a Linux PC that you can look at the logs?
Which model specifically of the tp-link? Can you do a
dmesg | grep usb
to get the discovery from the Raspbian, to see what driver is being used,
please?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

thecheeseontoast, 2020-04-03:
Hi, many thanks for your rapid answer!
It's a tp-link UE200
the dmesg response on raspbian was 

[    0.176190] usbcore: registered new interface driver usbfs
[    0.176297] usbcore: registered new interface driver hub
[    0.176527] usbcore: registered new device driver usb
[    0.475192] usbcore: registered new interface driver smsc95xx
[    0.503385] dwc_otg 20980000.usb: base=(ptrval)
[    0.906072] dwc_otg 20980000.usb: DWC OTG Controller
[    0.906171] dwc_otg 20980000.usb: new USB bus registered, assigned bus number 1
[    0.906244] dwc_otg 20980000.usb: irq 56, io mem 0x00000000
[    0.906715] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 4.19
[    0.906732] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.906744] usb usb1: Product: DWC OTG Controller
[    0.906754] usb usb1: Manufacturer: Linux 4.19.97+ dwc_otg_hcd
[    0.906764] usb usb1: SerialNumber: 20980000.usb
[    0.910140] usbcore: registered new interface driver usb-storage
[    0.916040] usbcore: registered new interface driver usbhid
[    0.916053] usbhid: USB HID core driver
[    1.339122] usb 1-1: new high-speed USB device number 2 using dwc_otg
[    1.581631] usb 1-1: New USB device found, idVendor=2357, idProduct=0602, bcdDevice=20.00
[    1.581659] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    1.581696] usb 1-1: Product: USB 10/100 LAN
[    1.581711] usb 1-1: Manufacturer: TP-LINK
[    1.581722] usb 1-1: SerialNumber: D0374587E190
[   19.132040] cdc_ether 1-1:2.0 eth0: register 'cdc_ether' at usb-20980000.usb-1, CDC Ethernet Device
[   19.132527] usbcore: registered new interface driver cdc_ether

Does that help at all?

starbasessd, 2020-04-03:
Which image are you using for motionEyeOS?
Try 20200203 for the Zero at
https://github.com/ccrisan/motioneyeos/releases/download/20200203/motioneyeos-raspberrypi-20200203.img.xz
Looking for the driver...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

thecheeseontoast, 2020-04-03:
Previously I was using 20190911, same issue for 20200203 image unfortunately.

starbasessd, 2020-04-03:
OK, I don't have a USB-Ethernet adapter handy.
I do have a USB WiFi Dongle (tp-link)
Installed the version of motionEyeOS to 4GB SDCard
Installed in RPiZero 1.3 (no built in WiFi)
Came straight up, discovered it, applied wpa_supplicant.conf, connected,
happy-hunky-dory.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

thecheeseontoast, 2020-04-03:
Ah unfortunately I have no USB WiFi Dongle handy.
I would be surprised if it's a driver issue as i read in #1960 
"motionEyeOS uses Raspbian's kernel (source tree as well as configuration). So whatever Raspbian supports should work with motionEyeOS as well"

starbasessd, 2020-04-03:
But it works in Raspbian?
Hmmm...
On the same PiZero?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

thecheeseontoast, 2020-04-03:
Indeed, no setup change asides from changing from Raspbian to Motioneyeos.
I can boot if I disable network. Is there a way of testing/configuring network/usb settings after disabling network in os.conf?

starbasessd, 2020-04-03:
Same OTG cable?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

thecheeseontoast, 2020-04-03:
Yes, same cable

starbasessd, 2020-04-03:
Yes, you could try treat it as a WiFi AP in the instructions I am
attaching, instead of wlan0 use eth0
Then connect it to a hub/switch, and connect a laptop or desktop to the
same switch, but not connected to the rest of the network.
An Ethernet crossover should work, too.
Detailed instructions are in the MotionEyeOS_AP.txt file
You may be able to safely leave out the hostapd.conf file
In dnsmasq.conf change wlan0 to eth0

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-04-03:
The Zero will give a PC all the settings it needs, you don't want to
conflict with your regular network...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

thecheeseontoast, 2020-04-03:
I will take a look and write back here my results. Many thanks for your help!

starbasessd, 2020-04-03:
I'm building a quick setup, pi3 with motionEyeOS setup and a Pi4 to connect
to it. If I find bad results or a better setup, I'll let you know, too.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-04-03:
That set up worked for me, so, you should be able to just use the
static_ip.conf and os.conf files and do what you need to be done.
Set yourself a static_ip.conf for your network (an address not in your
router's DHCP scope).
Set os.conf for networkless true
You should then be able to boot and access from any PC on the regular
network when connected via Ethernet.
The dnsmask.conf and hostapd.conf files can be left off the /boot
partition.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

thecheeseontoast, 2020-04-11:
Hi,
I eventually got this working via the workaround above. I have found the route of the issue however. The driver does not exist for the adapter I am using, or at least does not load, DESPITE the driver existing and working for raspbian. Why this is I do not know. I hope someone with far greater knowledge than I can figure things out. 
This is the tp-link UE200 ethernet adapter.
Many thanks to the help posted here!

starbasessd, 2020-04-11:
If the driver didn't exist in motionEyeOS the work-around wouldn't work,
anyway. I believe it's just the order that motionEyeOS looks for the
network before the driver module is enabled. This would be an @ccrisan
issue, I think.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ccrisan, 2020-04-12:
@thecheeseontoast we'll update the RPi kernel and firmware for a future prerelease, but first we need to do a stable release with all existing work.

jensenswitch, 2020-08-09:
I'm having the exact same problem, zero w but it looks for wired connection and panics when it cant find one..

starbasessd, 2020-08-09:
If you are using a USB Ethernet connection, you need to make sure you are
using one that is OTG compatible, or have an OTG adapter, and that it is
supported by Raspbian or Raspberry PI OS. I have 3, all can use the OTG
adapter, but 1 is not supported by Raspbian/RaspberryPiOS, and doesn't work
with motionEyeOS.
Also be sure power is going to the outer microUSB connector, and the OTG
adapter is going to the inner one.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?