motioneye-project/motioneyeos #2344: Some Menue Items ( File Storage, Motion detection, Text Overlay, etc) are missing

Alex81m, 2020-04-06:
Hi,
after new installation of motioneyeos on my Raspi 2 some menue items are gone.
For example the File Storage and Motion detection Items are missing. There is no advanced settings button. Also, I tried to turn on and off the Button „Fast network Camera“ without any success.
So I tried to use the dev-build also without success. Also, I changed the Raspi II to Raspi III with new image and a plain installation on raspian os without any apperance of the missing menue items.
Is this a bug or am I to blind to switch on the missing items?

### motionEyeOS Version

I am running motionEyeOS version: motionEyeOS dev20200223

### Board Model

I am using the following board/model Raspberry PI 2B  Raspberry PI 3B+

### Camera

I am using the following type of camera: Foscam 9903 Simple MJPEG Camera.

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue:
16 GB USB Pen-Drive

### Log Files

I consider the following log files relevant to this issue:
. motioneye.log and  boot.log
 
[image]
[boot.log](https://github.com/ccrisan/motioneyeos/files/4438033/boot.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/4438035/motioneye.log)

I highly appreciate your assistance.

starbasessd, 2020-04-06:
You are using the specific images for the specific Pis?
https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi2-dev20200223.img.xz
( Pi2 dev build)
or
https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi3-dev20200223.img.xz
( Pi3 dev build)

For example the File Storage and Motion detection Items are missing.
** What happens when you click the little blue triangle on the right of an
item?

There is no advanced settings button.
** That would be in the Expert Settings section, Click the little blue
triangle to the right of Expert Settings

Also, I tried to turn on and off the Button „Fast network Camera“ without
any success.
** That option is only for the CSI Pi Camera, doesn't work with USB or
Network cameras. It disables most features.
** If it is "On" this could also be your issue.

So I tried to use the dev-build also without success. Also, I changed the
Raspi II to Raspi III with new image and a plain installation on raspian os
without any apperance of the missing menue items.
** A lot of options are only available in motionEyeOS. If you run motionEye
on Raspbian, or Debian, or Ubuntu or other linux, you don't get / need the
other options.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Alex81m, 2020-04-08:
Hi,
I used both images 
For Rpi 2
https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi2-dev20200223.img.xz 
and for Rpi 3
https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi2-dev20200223.img.xz 

** What happens when you click the little blue triangle on the right of an item? T
See Screenshots below. Some Configuration Items are shown but not at all.

** What happens when you click the little blue triangle on the right of an item? 
There is no advanced settings button.
See Screenshot

** That option is only for the CSI Pi Camera, doesn't work with USB or Network cameras. It disables most features.
Well I read in an older Issue that this could also cause the issue that menue items are missing. In my case it is turned off
** A lot of options are only available in motionEyeOS. If you run motionEye on Raspbian, or Debian, or Ubuntu or other linux, you don't get / need the other options.
After the motioneEyeOS was not working as expected, I did the plain installation. As you can see in the screenshots I’m currently using MotionEyeOs.

[image]
[image]
[image]

starbasessd, 2020-04-08:
I'm hoping the wrong file for Pi3 is just a copy/paste error.
My Test setup:
PiZeorW, TIGA Generic USB Camera
Pi3B, Acting as Hub, connecting to 3 D-Link DCS-5020-L (Network Camera),
and motionEye Remote camera (PiZeroW)
Both running motionEyeOS 20200203
Both on Wireless
Attached here is screenshots from PiZeroW Configurations.
I'll send screenshots of Pi3B in a bit.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

macdroid53, 2020-05-15:
I am experiencing this exact issue with the Pi3 and Pi4 20200203 images.

beni9n, 2020-12-02:
I'm having the exact same issue with an RPi Zero W and a PiHut ZeroCam running meos-rpi-20200606. Have there been any updates?

starbasessd, 2020-12-02:
dev20201026 is working for me with PiCam v1.3 on RPiZeroW.
If you used other versions on the same IP Address and clicked 'remember me' when signing into admin, you may need to clear your browser cache.