motioneye-project/motioneyeos #2346: funcional in apps for see web cams

adsports, 2020-04-07:
functionality in apps for see web cams,
i trie whit HEDEN and E-VIEW7 , i inser ip public, port and user of motioeye but no see

thanks

starbasessd, 2020-04-07:
Could be blocked by the provider. Several US providers don't allow Port 25
or Port 80 connections through their firewalls.
I would connect a monitor and keyboard to the Pi, and see if it gets a
world IP address, or a 10.x.y.z, or a 192.168.x.y IP address, first, or see
if it is a continuous reboot situation.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

adsports, 2020-04-07:
when connected ip public for example. 203.187.XX.XX:8080 its ok on chrome, safari, etc. but APP for cameras no.

in the terminal PI how in ifconfig only 191.168.1.X local ip, but can see on internet on 205.305.XX.XX:8080

thanks

starbasessd, 2020-04-07:
Management port is either :80 (motionEyeOS) or :8765 (motionEye on
Raspbian, Ubuntu or other linux)
Streaming ports begin at :8081.
:8080 isn't used unless you made other changes.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-04-07:
> yes i change ports its works now , you can trie http://201.230.139.15:8081.
> its ok
> but this problem in app cell dont work
> [image: 92102808_806787226512459_1876013192014987264_n]
> <https://user-images.githubusercontent.com/23385694/78716139-49cfa900-78d3-11ea-9ddb-dfe9e53ece66.jpg>
> thanks

Does your web browser in your phone work, instead of rhe raspcam app? It
does for mine.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

adsports, 2020-04-07:
yes of course in the mobile browser yes, the doubt is for the apps

starbasessd, 2020-04-07:
Beyond the scope of my assistance, then, sorry. Good luck. Maybe others
here, or the App creators can help.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?