motioneye-project/motioneyeos #2347: Rpi 3b Camera LED

trosenda, 2020-04-07:
Hello,

I found this rather old thread https://github.com/ccrisan/motioneyeos/issues/1013

which tells for the Pi3 you have to "
Control via the Mailbox interface (/dev/vcio)
This C program can be used to control the camera LED."

Question is whether or not that was ever ported/compiled for motioneyeos ?

Thanks

starbasessd, 2020-04-07:
There is a button for the CSI camera LEDs. Haven't seen it work for
anything else.
Under Expert Settings

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

trosenda, 2020-04-07:
using this button requires reboot, I am looking for something more dynamic, to manually control an IR Cut.
but I found a suitable workaround (although it takes a second to switch) :
vcmailbox 0x00038041 8 8 134 0 to turn it off
vcmailbox 0x00038041 8 8 134 0 to turn it on

closed 2020-04-07 by trosenda