motioneye-project/motioneyeos #2348: Camera stops working (no web interface and ssh, stream froze) after couple of hours of running

Panjkrc, 2020-04-08:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200203

### Board Model

I am using the following board/model: Raspberry Pi Zero W

### Camera

I am using the following type of camera: MMAL Camera

My camera model is: MMAL Camera

### Network Connection

My motionEyeOS unit is connected to the network via: WIFI
### Peripherals

I am using the following peripherals that I consider relevant to this issue:

none

### Log Files

I consider the following log files relevant to this issue:

[dmesg.log](https://github.com/ccrisan/motioneyeos/files/4451825/dmesg.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/4451826/boot.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/4451829/messages.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/4451830/motioneye.log)

### Issue

Camera web interface stopped working after couple hours of powering it on, so did ssh connection, the stream froze (same picture was displayed). The camera didn't respond in any way so I had to restart it. Same happened couple of times so I installed version 20200203 hopping it would fix the issue (was running 20190911), but issue remained. I have no idea what could be the issue. Any help would be appreciated. Thanks in advance.

starbasessd, 2020-04-08:
If you have console or ssh access, can you run the command

vcgencmd measure_temp<enter>

after 30-60 minutes uptime and report the temperature, please?
What are the conditions when you notice it fails? Warm, cool, sunny?
Where is the PI? Indoors?
Good airflow or heatsinks?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Panjkrc, 2020-04-08:
I will run that command tomorow and come back to you, have to mount it back
It was about 10-15 °C when it failed. The pi is outside in a fake camera housing ( [this](https://www.ebay.com/itm/Outdoor-Solar-Power-CCTV-Camera-Fake-Security-Dummy-Surveillance-Red-LED-Light/113906881283?hash=item1a85611303:g:Vg8AAOSwjJddkM2O) one I think ), located under a roof in a shade. I put a heatsink on it ( the one I haven't been using for my rpi 3b+ anymore - [here](https://www.ebay.com/itm/4pcs-Aluminum-Heatsink-Radiator-Cooler-Kit-for-Raspberry-Pi-4B-with-StickNA/392752870664?hash=item5b71e53508:g:pZcAAOSwk75dOmcN)  )

Panjkrc, 2020-04-09:
So I tested it and I think the temperature might be a problem. When it was in shade temp was about 55 °C but when I left it in sun it was around 68-70 °C. Also the camera stopped working again after an hour. Should I put some vent holes in the case. I'm a bit worried that the moisture might mess with electronics.

starbasessd, 2020-04-09:
The Pi will try to protect itself when it starts getting warm, and will
shut down when hot. There are ways to cool, heat sinks, fans, heat pipes.
and you need to figure out what fits in your budget. Humidity shouldn't be
an issue, but actual moisture/wetness is. Vent holes can be protected
(spacer and cover on top). Fine mesh to keep bugs out, too.
Good luck.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Panjkrc, 2020-04-09:
Thank you for your response and advice!

closed 2020-04-09 by Panjkrc