motioneye-project/motioneyeos #2349: [BUG] Invalid Credentials

AlessioMeloni, 2020-04-08:
### Unable to login on the webUi after the activation of "monitoring option" as "user"

1. Login on the webUI as "user"
2. turn on the monitoring option

**Result:** From now on you will not be able to login on your webUI
**Fix:** Load a backup manually and reboot
**Problems with "read-only filesystem"?:**  `mount -o remount,rw /`

starbasessd, 2020-04-08:
Which version? 20200203 for PiZeroW, logged into system as user, I don't
even get the General section:
[image: image.png]

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

AlessioMeloni, 2020-04-09:
> Which version? 20200203 for PiZeroW, logged into system as user, I don't even get the General section: [image: image.png]

Version 20190911 for RaspberryPi3