motioneye-project/motioneyeos #2351: Recording location

jaimito2020, 2020-04-10:
Hi,

Congratulations on such amazing software, it really is quite fantastic!

I have been using the motioneyeos with no problems for many years to create remote cameras. I was initially using Ispy to do the recordings as I wanted to keep all recordings centralised. But I found the Ispy a bit buggy so decided to try another instance of motioneyeos running as a server so that all files would be centralised on that. 

So that would be a couple of remote motioneyeos cameras running over network but recording on the extra motioneyeos instance.

I can connect everything together with no problems as [adding remote motioneyeos camera]. The issue that I have is that on the server, I want to keep everything centralised but there doesn't seem to be an option to allow me to do that - I can only seem to record to the remote camera locations.

Is there a way to connect the remote motioneyeos cameras to a separate motioneyeos server so that all recordings are stored on the new motioneyeos server? I'm not sure if I'm missing something obvious!

Thanks in advance and thanks again for the incredible work!

Jaimito

starbasessd, 2020-04-10:
On the motionEyeOS hub, enable Samba Server, Authentication, and Write
Support under Services, click Apply and reboot.
On the motionEyeOS Cameras, change File Storage. Storage Device to Network
Share.
Enter the motionEyeOS Hub for Server
Share Name will be /storage
Share Username & Password will be the admin user and password for the Hub
Root Directory should be /<camera_name
Click Apply *** Very Important
Click Test Share
Repeat as needed on the other motionEyeOS Cameras
This will force the cameras to do their primary saving to the hub, instead
of the local camera.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jaimito2020, 2020-04-10:
Hi!

Wow thanks so much for getting back to me especially with such a comprehensive answer! That is amazing, I managed to get everything up and running with no problems by following your instructions.

I cannot thank  you enough! I wish you all the best and thanks again for making my life a bit easier!

Jaimito

starbasessd, 2020-04-10:
You're welcome. If completed, please close the ticket.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jaimito2020, 2020-04-10:
You're a legend - stay safe and thanks again :-)

closed 2020-04-10 by jaimito2020