motioneye-project/motioneyeos #2352: Raspberry Pi4: "Enable CSI Camera Led" does not switch Camera LED

ctschach, 2020-04-11:
### motionEyeOS Version

motionEye Version | 0.42
Motion Version | 4.2.2+gitUNKNOWN
OS Version | motionEyeOS dev20200223

### Board Model

Raspberry Pi 4B 2 GB

### Camera

Raspberry Pi MMAL camera

Using the Expert Settings Option "Enable CSI Camera Led" does not change the Camera LED on a Raspberry Pi 4.

Works fine with that camera on a RPi3, but does not work on a RPi4.

starbasessd, 2020-04-11:
Which PI camera (version#)?
Did you test with 20200203 or 20190911?
I don't have a spare CSI camera (on loan) for testing.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ctschach, 2020-04-13:
I used an IR Cut Camera which used the LED to switch on and off the IR filter (no automatic switching). It works fine on an RPi3B, but does not work on the RPi4.

As I mentioned above, I used the latest dev version (motionEyeOS dev20200223).

starbasessd, 2020-04-13:
Could you please test with 20200203, or 20190911? I would but I don't have
that particular camera.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jasaw, 2020-04-14:
Camera LED cannot be controlled on RPi4 as GPIO line has been removed: https://www.raspberrypi.org/forums/viewtopic.php?t=255568

starbasessd, 2020-04-14:
Noted, thanks for the info!

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?