motioneye-project/motioneyeos #2356: panic rebooting caused by network

adsports, 2020-04-16:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version motioneyeos-raspberrypi-20190911
[image]

### Board Model

raspberry pi Model B

### Camera

raspberry camera 5mpx

### Network Connection

ethernet and wifi adapter tenda

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

no

### Log Files

I consider the following log files relevant to this issue:

 only insert wpa_supplicant.conf
before firts iniciation
 

hi, i have raspberry pi mode B, when ethernet use its ok , but  when i insert wpa_supplicant.conf and connected adapter  wifi ,comming this error 

tranks for support

starbasessd, 2020-04-16:
Need to see boot log.
Which USB WiFi dongle are you using?
Does the WiFi dongle and wpa_supplicant.conf file work on Raspbian?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

adsports, 2020-04-16:
hi
which usb is. TENDA
dongle and wpa_suplicant works for 3 mins but is reboot

-Now I have another problem even though I restart my modem, it is configured like this
it shows me ip 10.0.0.157, but when testing it gets assigned 192.168.1.16 in the modem it shows as unknown device
-in ethernet is working
-how can I solve

extra fact
I have used the wpa_supplicant.conf file on raspberry pi3b +. and there are no problems

thanks
Atte
Jose Ruiz de castilla H.

starbasessd, 2020-04-16:
Boot log?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

adsports, 2020-04-16:
only screen
Is a log file generated inside the raspberry?
where can i find it

Atte
Jose Ruiz de castilla H.

adsports, 2020-04-17:
hi
that my modem assigns me as ip 10.0.0.XX gateway 10.0.0.1 and dns 10.0.0.1, how can I solve

It only happens with the wireless connection, I have tried on the raspberry 3b + and the wifi works well

thanks

Atte
Jose Ruiz de castilla H.

mathiasconradt, 2020-05-03:
Same issue with RPi B, Edimax Wifi Adapter (EW-7811Un)

[image]

starbasessd, 2020-05-03:
Which RPi are you using? Original Pi (Pi1B)? Pi Zero, ZeroW?

Try motionEyeOS from:
20200203
https://github.com/ccrisan/motioneyeos/releases/download/20200203/motioneyeos-raspberrypi-20200203.img.xz

It appears you may be using a PiZero or Pi1.
If the Zero, are you sure you have an OTG USB adapter cable?
Power plugged into the USB port farther away from the HDMI port? WiFi
dongle plugged in via OTG cable into nearest port to HDMI connector?

Did you put a wpa_supplicant.conf file in the /boot partition?
It should look something like:
-----
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
    ssid="<your_ssid>"
    psk="<your_password>"
key_mgmt=WPA-PSK
}
-----
Another test:
Install raspbian  on the SDCard, and see if the WiFi dongle is discovered.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

towfeeqf, 2021-02-01:
I had a similar issue. It was related to file extension.

You may click on the comment to see in detail.

https://github.com/ccrisan/motioneyeos/issues/2537#issuecomment-770477111

closed 2022-01-19 by starbasessd