motioneye-project/motioneyeos #2357: Time offset live image - MotionEyeOs image

JBSullivan, 2020-04-18:
Hello, I am using MotionEyeOS (4.2.2.) with a Raspberry 3b+.
I have 5 network (2 x LAN - 3 x Wlan) cameras connected there. The settings for all 5 cameras are identical.

Here are the settings: 640x480 / 2 Frames
Only the "Video Device" option is enabled - all other settings are disabled.

I have the problem that between the realtime image and the images MotionEyeOS send via streaming URL to my home automation system is sometimes several minutes delayed.

Is this a performance problem, even though I have already turned the settings down completely?
Is there somthing i can do to eleminate this problem?

Here is the result from TOP

Mem: 360096K used, 405500K free, 168K shrd, 14740K buff, 40256K cached
CPU: 93% usr 1% sys 0% nic 4% idle 0% io 0% irq 1% sirq
Load average: 4.89 5.26 5.26 8/131 1369
PID PPID USER STAT VSZ %VSZ %CPU COMMAND
3979 1131 root S 418m 56% 93% /usr/bin/motion -n -c /data/etc/motion
1131 1 root R 54216 7% 3% /usr/bin/python /usr/lib/python2.7/sit
1831 1085 root S 5644 1% 0% sshd: admin@pts/0
1849 1835 root R 2592 0% 0% top
9 2 root SW 0 0% 0% [ksoftirqd/0]
26 2 root SW 0 0% 0% [ksoftirqd/3]
838 2 root IW 0 0% 0% [kworker/0:0-eve]
1086 2 root IW 0 0% 0% [kworker/3:1-eve]
1166 1131 root S 23020 3% 0% /usr/bin/python /usr/lib/python2.7/sit
1035 1 root S 8180 1% 0% /usr/sbin/ntpd -g -c /data/etc/ntp.con
1085 1 root S 5644 1% 0% /usr/sbin/sshd -f /var/run/ssh/sshd_co
291 1 root S 4216 1% 0% /sbin/udevd --daemon --resolve-names=n
966 1 root S 3804 0% 0% dhclient -cf /var/cache/dhclient.conf
1835 1831 root S 3580 0% 0% -sh
1178 1 root S 3320 0% 0% {S85motioneye} /bin/bash /etc/init.d/S
265 1 root S 3320 0% 0% {S11throttlewatc} /bin/bash /etc/init.
874 1 root S 2832 0% 0% /usr/sbin/rngd
1050 1 root S 2592 0% 0% /usr/sbin/crond -c /etc/crontabs
1 0 root S 2464 0% 0% init
229 1 root S 2464 0% 0% /sbin/syslogd

starbasessd, 2020-04-18:
OK. Which image version are you using of motionEyeOS?
Are you seeing the long delay at the motionEyeOS machine, or at the home
automation?
Are you connecting to the motionEyeOS streaming ports? 8081, 8082, 8083,
8084, 8084, individually, or does the Home Automation device automatically
figure it out?
We had a similar issue recently, and it was something on the Home
Automation software/plugin side.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

JBSullivan, 2020-04-18:
OK. Which image version are you using of motionEyeOS? - **as i wrote 4.2.2.+gitUNKONWN downloaded direct form here**

Are you seeing the long delay at the motionEyeOS machine, or at the home
automation? **The delay is on both the same.**

Are you connecting to the motionEyeOS streaming ports? 8081, 8082, 8083,
8084, 8084, individually, or does the Home Automation device automatically
figure it out?
**Each camera use the streaming port MotionEyeOs is recommended - from 8081 Cam1 - to 8085 Cam5**

We had a similar issue recently, and it was something on the Home
Automation software/plugin side.

**I use ioBroker, but as i wrote, the delay is the same on the MotionEyeOs Web interface site.**

starbasessd, 2020-04-18:
Grab the image from here for your particular PI:
https://github.com/ccrisan/motioneyeos/releases/tag/20200203
and re-image.

Can you remove all but 1 camera (preferably keeping one that is showing the
delay) and test?

Also, does the delay happen on the camera itself or is it real time?

Which Camera(s) are you using?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

JBSullivan, 2020-04-19:
I uploaded the new image and except for one camera I didn't create all the others.

Nevertheless I have a 3 minute time offset between the image which is delivered by the original camera stream and the image which is played back in MotioneyeOS.

SD Card is - SanDisk Extreme 64Gb UHS class 3 ( 30MB/s)

Original camera 13:33
System Time 13:33
MotionEyeOS camera image but is from 13:30

I use different Chinese cameras from different manufacturers.

[image]

starbasessd, 2020-04-19:
If you turn on Text Overlay, (leave at default) which time date does it
follow? The Camera's or the local computer you took the screen capture from?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

JBSullivan, 2020-04-19:
MotionEyeOS follows his "own time" :(

Picture Time = Camera Time
Text Overlay = MotionEyeOS
Calender = local PC

Screenshot was from the local Computer in the same network like MotionEyeoS is.

[image]

starbasessd, 2020-04-19:
OK.
To confirm.
If you go out to the camera, and it captures your presence, it does not
show up in the stream until 3 minutes later, correct?
Or are you just seeing the 3 minute difference in the timestamps?
Is your camera connected via USB or CSI or wifi/ethernet?
If, in Expert Settings, can you change
Date Method to NTP
NTP Server to poo.ntp.org
Date Updating Timeout to 10 seconds
And let me know which clock it follows?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

JBSullivan, 2020-04-19:
> OK. To confirm. If you go out to the camera, and it captures your presence, it does not show up in the stream until 3 minutes later, correct? 
**YES - correct**

Or are you just seeing the 3 minute difference in the timestamps?
**No, this is real. Example: If the camera image shows a human being at 1:30 pm, MotionEyeOS will also show me 1:30 pm - but I will not see this 1:30 pm display until the system time and the camera's own time are already 1:33 pm - it is a video recorder with negative time shift**

 Is your camera connected via USB or CSI or wifi/ethernet? 
**Two cameras are via LAN, 3 cameras via Wlan in the same network where the Raspi with MotionEyeOS (LAN) is running.**

If, in Expert Settings, can you change Date Method to NTP NTP Server to poo.ntp.org Date Updating Timeout to 10 seconds 
**These settings have always been / are exactly the same for me.**

And let me know which clock it follows? **-- see article before.**

starbasessd, 2020-04-19:
@jasaw or @ccrisan/motioneyeos <motioneyeos@noreply.github.com>  any
thoughts?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?