motioneye-project/motioneyeos #2360: FTP is trying to access path with //

uzelessknowledge, 2020-04-20:
I am trying to configure FTP on my Pi zero w with MotionEyeOS. I continue to see Authorization Failed. I looked in the log and I can see that MotionEyeOS is inserting 2 backslashes in the path. 
ERROR: sftp: cURL upload failed on sftp://192.168.2.100:22/users/travis/public/motioneye/camera1//motioneye_test/1587353044.13: Authentication failure

What can I do to use the correct FTP path? Right now I have the FTP server setup on my Mac. My plan is to use this to take time-lapse of a construction project. 
[motioneye (4).log](https://github.com/ccrisan/motioneyeos/files/4501112/motioneye.4.log)

### motionEyeOS Version

I am running motionEyeOS version: 20190911

### Board Model

I am using the following board/model: Pi Zero w (also tried with Pi3)

### Camera

I am using the following type of camera: V4L2

My camera model is: Miuzei Raspberry Pi 3 Camera Module, 5 Megapixels Wide-Angel Fish-Eye Camera, OV5647 Sensor with Flex Cable for Raspberry Pi 4/ 3B+/ 3B/ 2B/ Pi Zero
https://www.amazon.com/gp/product/B07BTP8VXL/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi
### Log Files

I consider the following log files relevant to this issue: motioneye.log

starbasessd, 2020-04-20:
It's trying to connect to SFTP, not FTP according to the error message.
SFTP is SSH/SCP encrypted.
Turn on Upload Media Files which enables FTP uploads
[image: image.png]

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

uzelessknowledge, 2020-04-20:
Hmm, Maybe I don't have my mac set up correctly. I can't FTP to it with an FTP client, but I can SFTP to it.
Here are the settings in MotionEyeOS and an updated log. 
[image]
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/4504361/motioneye.log)

ccrisan, 2020-04-20:
SFTP is not FTP. Use the "SFTP Server" option to upload to an SFTP.

closed 2020-04-20 by ccrisan

uzelessknowledge, 2020-04-20:
@ccrisan Correct. I showed this after @starbasessd said to use FTP. 

When I use SFTP I see double backslash in the logs after the folder that is being accessed. 
sftp://192.168.2.100:22/users/travis/public/motioneye/**camera1//motioneye
[motioneye-2.log](https://github.com/ccrisan/motioneyeos/files/4505193/motioneye-2.log)
**_test/1587353044.13: Authentication failure

starbasessd, 2020-04-20:
Your original post: " I am trying to configure FTP on my Pi zero w with
MotionEyeOS"
I pointed out the log showed it was trying to connect to SFTP
I pointed out where you select where and what protocol you use to upload
For Server, you only put in the IP Address or the name of the server.
Put the username, password & path in the other appropriate boxes.
Select apply before trying the test button.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

uzelessknowledge, 2020-04-20:
Hey @starbasessd, it was my fault for saying FTP when I was trying SFTP. 
I’ve tried both and can’t connect to my Mac either way. I know SFTP works because I can use a client, FileZilla, and connect. 

Yup, I entered the info and clicked apply before testing the connection. 

I was able to use Local Share to save the images and was able to save them to my Mac. So I’m good there. 

Still not sure why the SFTP is showing double backslash in the logs. 

I appreciate your help and any help anyone can provide. Apologies if my last comment came across the wrong way.

ccrisan, 2020-04-20:
@uzelessknowledge understood. Well, it probably has to do with the way you entered the location. However an extra slash shouldn't affect the way Linux interprets paths.

uzelessknowledge, 2020-04-20:
OK, I've made progress. Googling helped me figure out that the user was not added to the SACL, by Mac OS. I added them through terminal. Whew!
I can connect FTP to the Mac now. Now I'm getting file exists. I've created a directory on my mac, in my public folder. I've given it permission. What file is it telling me already exists?

2020-04-20 18:02:11: [motioneye]    DEBUG: ftp: creating connection to motioneye@192.168.2.100:21
2020-04-20 18:02:11: [motioneye]    DEBUG: 200 HEAD / (127.0.0.1) 61.16ms
2020-04-20 18:02:11: [motioneye]    DEBUG: ftp: ensuring path /Users/travis/public/motioneye
2020-04-20 18:02:11: [motioneye]    ERROR: ftp: 550 travis: File exists.

uzelessknowledge, 2020-04-20:
Ok, I can't use public for some reason. I created a folder under the Motioneye user, but not in public. Gave it permission and BAM! I can connect to it. File is saved to FTP!

Now if I can figure out how to use motion detection and only take 1 shot when motion, wait a period of time and then if more motion take a shot. I think this is Motion Gap mixed with Single Image Motion Triggered (one picture), but it seems to take continuous shots if there is continuous motion.