motioneye-project/motioneyeos #2361: Store records only to Google Drive

tuga49, 2020-04-21:
Hi, 
I'm storing records to Google Drive, however, tey still be stored in my local storage. Is possible to disable it from storing in local and keep only Google Drive? thank's

starbasessd, 2020-04-21:
Unfortunately, no. You can re-direct it to a local storage (Network/SMB)
but the way it works it creates the first file In File Storage, then copies
it to Uploads.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tuga49, 2020-04-21:
> Unfortunately, no. You can re-direct it to a local storage (Network/SMB) but the way it works it creates the first file In File Storage, then copies it to Uploads.
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

ok thanks. and it's possible to write on local storage and delete after 1h or 1day? but keep on drive for 1 month for example?

starbasessd, 2020-04-21:
Several ways to do it
write a script to simply delete everything in the folder after each picture
or movie triggered by a "Run Command" with a delay built in or as a cron
job every so often.
I am not sitting at home so can't look at it right now, but there is a
trigger for Google drive to clean cloud too, you want that optio0n off, if
you turn on the 1 hour or day or week or whatever option.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tuga49, 2020-04-21:
> Several ways to do it write a script to simply delete everything in the folder after each picture or movie triggered by a "Run Command" with a delay built in or as a cron job every so often. I am not sitting at home so can't look at it right now, but there is a trigger for Google drive to clean cloud too, you want that optio0n off, if you turn on the 1 hour or day or week or whatever option.
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

So what I wan't, is keep files in google drive for some time. it can be 1 month or 1 week. I have 5TB storage, so theres no problem. But in local, just keep the minimum time so it dont fill up my local memory. example, delete all days at 12PM or so. Tha'ts easy to do?

starbasessd, 2020-04-21:
Yes. Turn off the Clear Cloud option and set your Preserve Pictures and
Preserve Movies options to however long you want.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tuga49, 2020-04-21:
> Yes. Turn off the Clear Cloud option and set your Preserve Pictures and Preserve Movies options to however long you want.
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

oh. thants a very good workaround. I can delete drive folder manually everymonth or find a way to do that automatically. but for now I think it's ok :) thanks for your help bro ;)

lcharles123, 2020-05-06:
If you have enough RAM memory, you can create a little disk entirely on RAM, so you can store, upload and delete. It is useful to prevent wear of flash memory, if you are using it to recording