motioneye-project/motioneyeos #2362: [QUESTION] Pass though video stream to web client without re-encoding?

AndreVallestero, 2020-04-23:
Does MotionEye do a simple pass-through of the cam video-stream if quality and framerate are maxed? What about if framerate is lowered? If motion optimization is on? Ideally, I would like MotionEye to not re-encode the originally ingested stream and rather pass it through when streaming it to the web client. Obviously this is not possible at lower quality settings since it would have to re-encode for a lower bitrate target, but I'm not sure if frame rate would also affect this since h264 bitstreams are reliant on keyframes and frame deltas which would change based on framerate.

My goal is to reduce CPU/VPU usage to a minimum by avoiding re-encoding since bandwidth is not an issue for me.

mitek96, 2020-05-20:
I have the same question. In my case streaming fps is low(15 fps), but recording fps is ok (25 fps). Is there an option to pass through stream?

kpcz, 2022-09-16:
Same question, any ideas how to pass through stream without encoding and using lot of CPU power?

starbasessd, 2022-09-16:
For that particular scenario, you can test various settings in Settings, Video Streaming. Try quality 0 or quality 100, Turn it off all together, etc. Since motionEye is a wrapper to be a monitoring system for motion, I doubt that would be possible. 
Here are the basic settings for the [Livestream](https://motion-project.github.io/motion_config.html#OptDetail_Stream)
I would consider putting such a feature request in their github (motion-project is a totally different group and project)
at their contacts pages: https://github.com/Motion-Project/motion