motioneye-project/motioneyeos #2364: garden camera for House Wren

joeydream, 2020-04-24:
Thanks to MotioneyeOS and ccrisan, as title I installed a system of RPi in my garden this spring. I would like to share my happiness with you all and ask a question at the end

[image]

hardware details:
- RPi zero w with pi cam
- RPi 3B for overall control: record movie and uploaded movie to NAS
remarks: detect motion, 15s video record and upload

my questions: 
how to improve the movie by increasing the fps, now is 20fps but seen lag too much.

starbasessd, 2020-04-24:
On the PiZero turn on Fast Network Cam under Expert Settings
On the Pi3B you can try doing pass through, or one of the other file
formats. Turn of still images.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

joeydream, 2020-04-24:
> On the PiZero turn on Fast Network Cam under Expert Settings On the Pi3B you can try doing pass through, or one of the other file formats. Turn of still images.

FastNetworkCam has turn ON, 
passthrough, okay i will turn on it (but what's it mean, i want to understand more)
turn off the still images, okay

anyway, thx

starbasessd, 2020-04-24:
Fast Network Cam turns off a lot of features and just streams the actual
video.
Passthrough saves the video without (much) extra processing on the 3B
Depends on which is more what you want. Some formats are designed for
streaming, some are for more/less compression, etc.
Looks like you just wanted the movies, so no need to waste processing time
on stills.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?